black-parrot/black-parrot #236: D$ LRU logic sends 'X as lruWay in LCE Request

muwyse, 2019-08-08:
The LRU logic in the D$ (and likely I$) may not be functioning properly in situations where the a cache set is filled with valid blocks in every way. The D$ appears to be sending 'X as the lruWay field in the LCE Request message, instead of providing a valid way to use for replacement.

branch: dev_lrux
Steps to reproduce: rebuild demo test programs, rebuild CCE microcode roms, run test from top
1. from bp_common/test: make clean; make demos
2. from bp_me/src/asm: make clean; make roms
3. from bp_top/syn: make build.v sim.v CFG=e_bp_dual_core_cfg PROG=write_demo_2 CCE_TRACE_P=1 DUMP=1

farzamgl, 2019-08-08:
I can't make the demos on this branch. Are the test Makefiles outdated?

muwyse, 2019-08-09:
I don't think so. The branch referenced is a single commit ahead of dev (as of yesterday). You might need to explicitly do 'make perch' or something similar to build the perch stuff in bp_common/test

dpetrisko, 2019-08-09:
I think that could be the issue. If you do all_mem, libperch is automatically built, but with just demos it’s possible it’s not

dpetrisko, 2019-08-09:
So it looks like the LRU straight up isn't working.  The D$ is configured to use invalid ways before checking the LRU (which is always X), so when all ways are full it's just X.  Looking into it

farzamgl, 2019-08-09:
Are the IO to the LRU enc/dec modules X too?
Because we are using the same modules for TLB and they seem to be working there.

dpetrisko, 2019-08-09:
It's not the lru modules themselves, I think the stat_mem access is misaligned with the lce_req

dpetrisko, 2019-08-09:
(at least, I don't know if it's the LRU module.  It's X-in X-out at the moment)

dpetrisko, 2019-08-09:
Looks like it's writing to the stat_mem instead of reading, when trying to get the LRU.

dpetrisko, 2019-08-09:
Oh, I think it's due to an LR/SC sequence.

dpetrisko, 2019-08-09:
Fixed the bug, no more Xs in LRU field.  This program still doesn't terminate, though...

dpetrisko, 2019-08-09:
Fixed completely.  Our barrier implementation is buggy, doesn't terminate.  Substituted the bp_finish instead of bp_barrier, now program terminates.  Merging in fix and closing issue

closed 2019-08-09 by dpetrisko