black-parrot/black-parrot #245: Please add actual license

mithro, 2019-09-03:
The lack of actual LICENSE on black-parrot is preventing the project be used as a benchmark or tests by a number of other projects.

Please release it under an OSI license ASAP.

taylor-bsg, 2019-09-03:
Thank you for your comment. Certainly using BP as an input test case is a
permitted use of the code.

Regarding the license, we have been focused on reaching V 1.0 of
BlackParrot which means a reasonably well-tested silicon-validated
Linux-capable RISC-V multicore. We taped out a 4-core BP in 14nm two months
ago.

At the time of that release — most likely in a  month — we will select the
finalized license.  We are open to your suggestions. We are deliberating
between BSD, Apache, and SolderPad.

mithro, 2019-09-03:
I highly recommend Apache 2.0 as your best choice.

taylor-bsg, 2019-09-03:
Can you explain why?

mithro, 2019-09-03:
* Is permissive (same as MIT, BSD-2)
 * Is wordier so there is less room for misinterpretation
 * Contains explicit patent language (see also https://hackernoon.com/4-lessons-from-the-react-patent-license-controversy-3da3c4baf3a4)
 * Wide compatibility with other licenses

dpetrisko, 2019-10-02:
Thanks for your patience on this issue.

BlackParrot is now licensed under BSD-3.  Let us know if you run into any trouble using BlackParrot as a benchmark!

closed 2019-10-02 by dpetrisko