black-parrot/black-parrot #255: TLB should be single-ported

dpetrisko, 2019-09-17:
https://github.com/black-parrot/pre-alpha-release/blob/master/bp_be/src/v/bp_be_mem/bp_be_dtlb/bp_be_dtlb.v

The TLB is effectively 1rw at the moment.  Having r_ready depend on ~w_v_i can cause combinatorial loops, if the read _cmd->tlb depends on ready.

Current interface:
```
 (input                               clk_i
  , input                             reset_i
  , input                             flush_i

  // Connections to Cache
  , input                             r_v_i
  , output                            r_ready_o
  , input [vtag_width_p-1:0]          r_vtag_i

  , output logic                      r_v_o
  , output logic [entry_width_lp-1:0] r_entry_o

  // Connections to PTW
  , input                             w_v_i
  , input [vtag_width_p-1:0]          w_vtag_i
  , input [entry_width_lp-1:0]        w_entry_i

  , output logic                      miss_v_o
  , output logic [vtag_width_p-1:0]   miss_vtag_o
 );
```

Proposed interface:
```
 (input                               clk_i
  , input                             reset_i
  , input                             flush_i

  // Connections to Cache
  , input                                     v_i
  , input                                     w_i
  , input [vtag_width_p-1:0]   vtag_i
  , input [entry_width_lp-1:0]  entry_i

  , output logic                                    v_o
  , output logic                                    miss_v_o
  , output logic [vtag_width_p-1:0]   miss_vtag_o
  , output logic [entry_width_lp-1:0] r_entry_o
```

Another thought is to move the TLB to bp_common, since it is shared between FE and BE.  Currently we have duplication a weird FE->BE dependency

What do you think?

dpetrisko, 2019-09-17:
Oh, another note.  By construction, we'll never need simultaneous RW on the TLB anyway (we do a PTW before a write, which stalls both the FE/BE).  So this is strictly clarity, there should be no performance impact.

taylor-bsg, 2019-09-17:
By the way the signals are incorrectly named if a ready_o depends on a
valid_i — this is a violation of ready/valid handshaking.

farzamgl, 2019-09-17:
#256

closed 2019-09-25 by dpetrisko