black-parrot/black-parrot #267: Guarantee forward progress on LR/SC

dpetrisko, 2019-10-02:
The following change was proposed on riscv-isa-dev:

> During the public review period for ratification of the base ISA and standard extensions, the concern was raised that the LR/SC forward-progress guarantee is too onerous to provide in some microarchitectures.  The Board of Directors opted to postpone ratifying the A standard extension until this matter could be studied further.
> 
> The memory model task group has since proposed a weaker LR/SC forward-progress guarantee.  It is attached to this email, and we are opening it up for public comment for 45 days.  Note that only the new LR/SC semantics are up for review; the rest of the A extension (all aspects of AMOs, plus the LR/SC instruction encoding) already had a successful public review.  Please provide feedback via the GitHub issue tracker using the Base ISA Ratification tag (https://github.com/riscv/riscv-isa-manual/labels/Base%20ISA%20Ratification), or, for the GitHub-disinclined, you can reply to this email instead.
> 
> Informally, the new scheme now only guarantees that _some_ hart will make progress with respect to LR/SC, instead of all harts.  Put another way, only freedom from LR/SC livelock is assured, whereas before, the hardware was also responsible for freedom from starvation.
> 
> This new formulation is more in line with other architectures' LR/SC primitives, but is a bit more explicit about what software can expect.  It remains strong enough to implement, among other things, the C++ language standard.
> 
> Existing hardware that correctly implements the old scheme remains conformant under this new proposal.
> 
> 
> Andrew, on behalf of the memory model task group

Proposal: We can guarantee forward progress during LR/SC by ignoring invalidations to that line until the atomic transaction is complete.  We probably should only lock the line after N failures and automatically drop the reservation after M cycles.  Possibly we can reuse the same counter as the livelock LCE counter in dcache.v?

Verification: A stress test program for multicore contention where each core constantly LR/SCs.  If it reaches the end of the program, forward progress is being made.  Of course, this doesn't formally verify correctness, just efficacy, so further testing may be needed.

PPA impact: Minimal.  Providing a strong software guarantee is useful though

taylor-bsg, 2019-10-03:
Just a general note that counters are discouraged if possible.  In general,
we want BP to "fail fast", because a bug that happens every billion cycles
is a lot easier to find than one that happens every trillion cycles. =)

M

dpetrisko, 2019-10-03:
Makes sense.  Is there an alternative here though?  If we lock the line, we have to unlock it somehow.  That's either based on time or number of blocked requests.  Since we don't have an easy way of measuring backpressure from the LCE, time is simpler.

taylor-bsg, 2019-10-03:
Not reading the spec super carefully here, but I would say, something like,
it auto-unlocks after the commit of 1) an SC, 2) control flow transfer or
3) 16 instructions.

dpetrisko, 2020-01-23:
Included in latest dev.  Conditions are:
1) Lock timeout starts when cache miss OR LR
2) Lock timeout ends at any hit (includes SC, load or store) or (configurable) N cycles

This is compatible with the spec and works for our rollback-based miss system

closed 2020-01-23 by dpetrisko