black-parrot/black-parrot #269: Prerequisites for Ubuntu (byacc)

scanakci, 2019-10-09:
byacc is a prerequisite for Ubuntu system. It would be good to update the getting started.

dpetrisko, 2019-10-09:
Thanks! We haven’t tried getting started on Ubuntu yet, let us know anything else that comes up!

dpetrisko, 2019-10-09:
Do you know which dependency byacc is used for?

scanakci, 2019-10-09:
I did not log the full error but I have this error in my records : "pre-alpha-release/external/Makefile.tools:62: recipe for target 'cmurphi' failed"

Hope it helps

dpetrisko, 2019-10-09:
@muwyse Are you using CMurphi for an ME regression? Should we be :)?  Otherwise, maybe we can put cmurphi under a different make tools target.

muwyse, 2019-10-09:
It is not used in regressions; it is used for coherence protocol verification, and the protocol should be fixed at this point.

dpetrisko, 2020-03-05:
Added prereq to latest dev

closed 2020-03-05 by dpetrisko