black-parrot/black-parrot #273: Integrate google riscv-dv

taylor-bsg, 2019-10-15:
https://github.com/google/riscv-dv?files=1

dpetrisko, 2019-10-15:
Yep, it's been on the docket for a while: #133 .  It's improved a ton in the last 6 months.  I'm guessing you have your eye on the CSR testing tool -- probably a great idea.

dpetrisko, 2019-10-15:
https://github.com/lowRISC/ibex/blob/master/doc/verification.rst

This is an example of IBEX doing verification vs Spike.  It looks great, we should adopt something similar

taylor-bsg, 2019-10-15:
Yes it occurs to me that the turn around time issues with long running
tests are also solvable by finding shorter tests of the same functionality
through random program generation. We still need to have a way of debugging
bugs-after-a-trillion-cycles, but we can decrease their frequency with this.

dpetrisko, 2019-10-18:
Started integration here: sw_dev_project_googledv

It's a little finicky.  Basically, I can only get the basic RV64I tests working.  When I try to generate the MMU/privilege/CSR stress tests, I get errors about unsolvable constraints.  AFAIK, riscv-dv has only been tested thoroughly with a small number of cores, so it's possible that there are some glitched assumptions about core feature sets.

dpetrisko, 2020-12-22:
Integrated for a while

closed 2020-12-22 by dpetrisko