black-parrot/black-parrot #291: Differentiate between CCE->L2 IF, IO network and MEM network

dpetrisko, 2019-12-05:
Currently, all three of these function as our command / response networks.  However, this is very wasteful because we require 500+ bit buffers for I/O and MEM adapters, even though the size of the requests is generally at most 64.

We need cacheline sized transfers between CCE->L2, but not anywhere else.  This may also allow us to get rid of wasteful unused payloads being sent for things like L2->DRAM

taylor-bsg, 2019-12-06:
Even in CCE->L2, there is a benefit of not doing a wide cacheline size at a
time. The DRAMs return critical word first, so if we transfer it in flits,
we don't have to wait for the last word before sending it on.

Another thing that might make sense to refactor is the L1 cache. It
currently expends a lot of effort in order to be able to send out an entire
cache line as a wide word, but this is not a necessity.

M

dpetrisko, 2019-12-06:
That’s an interesting point. We should consider a narrow interface if we do split between the $ and LCE, like we spitballed for “BP_minimal”. It would basically change it from data_mem_pkt & v to data_mem_pkt & mask, or possibly a similar dma interface to L2.

dpetrisko, 2019-12-06:
I agree that a wide interface between CCE and L2 doesn’t necessarily make sense in the context of wormhole routing

dpetrisko, 2020-01-23:
#302.  We can add network enhancements / L1 refactor to a new issue when they arise

closed 2020-02-04 by dpetrisko