black-parrot/black-parrot #294: Critical path: FE I$ to next pc

dpetrisko, 2020-01-06:
Critical path is I$ tag compare to next PC.  Specifically, it's I$ tag compare->data mux->(instruction)->instruction scan->(is branch)->pc+branch offset->next pc.

Possible solutions:
1) Add a cycle for branch override. This will complicate the control logic
2) Pre-decode instruction type and store in I$ data. This is more complicated cache logic 
3) Do not override un-predicted branches.  Instead, store in BTB upon predicted taken, then branch resolution will handle it.  This shouldn't be a huge performance hit, because misprediction penalty is only 2 cycles larger than branch taken penalty, and we only pay this on branches not in the BTB.

taylor-bsg, 2020-01-06:
Pretty common path. We have a standard suite of tricks in BSG for
optimizing front ends. After we get some more benchmarks up and running we
can evaluate tradeoffs.

M

dpetrisko, 2021-03-16:
Need to reevaluate

closed 2021-03-16 by dpetrisko