black-parrot/black-parrot #295: Syntax errors in some files?

ptpan, 2020-01-07:
I was trying to run verilator simulation on the current master branch. Verilator reported several syntax errors in the following files: 
```
%Error: /black-parrot/bp_me/src/include/v/bp_cce_inst.vh:1081: syntax error, unexpected <<, expecting TYPE-IDENTIFIER
%Error: /black-parrot/bp_me/src/include/v/bp_cce_inst.vh:1084: syntax error, unexpected ===
%Error: /black-parrot/bp_me/src/include/v/bp_cce_inst.vh:1095: syntax error, unexpected '}'
%Error: /black-parrot/bp_me/src/v/cce/bp_cce_msg_cached.v:319: syntax error, unexpected <<
%Error: /black-parrot/bp_me/src/v/cce/bp_cce_msg_cached.v:506: syntax error, unexpected '=', expecting IDENTIFIER
%Error: /black-parrot/bp_me/src/v/cce/bp_cce_msg_cached.v:522: syntax error, unexpected '=', expecting IDENTIFIER
%Error: /black-parrot/bp_me/src/v/cce/bp_cce_msg_cached.v:523: syntax error, unexpected '=', expecting IDENTIFIER
...
```

Seems like some conflict markers accidentally got committed to these files?

dpetrisko, 2020-01-07:
Thanks for the report! Yep, you’re catching me currently updating the repo and fixing some weird merge conflicts. I’ll close this issue when resolved.

dpetrisko, 2020-01-08:
Hi, build issues should be resolved.  Sorry for the inconvenience!

ptpan, 2020-01-08:
Thanks for the fix Dan!

closed 2020-01-08 by ptpan