black-parrot/black-parrot #298: Add error codes to host interface

dpetrisko, 2020-01-19:
Currently, when we drop down into firmware for illegal instructions, we decode the instruction and if we have an emulation routine, jump to it.

However, this makes it difficult to debug the cause of program hangs.

Instead, we should send a packet to host to signal an error.  The host can then use the MMIO debug bus to inspect the state of the processor and print the error to a user.

taylor-bsg, 2020-01-21:
Sometimes illegal instructions are part of the legal execution of the
processor. In fact in the x86 BIOS there were illegal instructions that had
to be serviced in order to continue booting, as we found out in VirtualPC.
I think at the very least we should fire an assert in the Verilog that has
some detailed information (physical and virtual address, instruction
encoding?)

m

dpetrisko, 2020-01-21:
>serviced in order to continue booting

serviced by what? I'm talking about the case where the firmware doesn't know how to handle the instruction.  

>I think at the very least we should fire an assert in the Verilog that has
some detailed information (physical and virtual address, instruction
encoding?)

There's not a verilog assert for this, because illegal instruction servicing as part of emulated instructions, virtualization, etc. is perfectly legal at the hardware level. It's only at the firmware level that there is a detectable problem. This is basically unhandleable trap / NMI firmware support.  We could even add a watchdog timer here, if we think it's useful

taylor-bsg, 2020-01-21:
> serviced in order to continue booting
>
> serviced by what? I'm talking about the case where the firmware doesn't
> know how to handle the instruction.
Hm .. maybe we need to be more precise.  Linux will certainly have an
illegal instruction trap handler, and even the applications can potentially
register a handler for the POSIX illegal instruction signal. So in most
scenarios, I foresee that the firmware will pass this on to the OS and then
the OS will forward it down the chain.

> I think at the very least we should fire an assert in the Verilog that has
> some detailed information (physical and virtual address, instruction
> encoding?)
>
> There's not a verilog assert for this, [...]
The assert should fire in cases where it is sufficiently unusual to
indicate that something may be going wrong ( I think this was the original
intent of the github issue?)

dpetrisko, 2020-01-21:
> Hm .. maybe we need to be more precise.  Linux will certainly have an
> illegal instruction trap handler, and even the applications can potentially
> register a handler for the POSIX illegal instruction [...]

Maybe @farzamgl  can weigh in here, but I believe the sequence is:
1) firmware sets up illegal instruction delegation in medeleg to linux trap handler.
2) illegal instruction happens in U-mode
3) trap to M-mode
4) M-mode inspects, can firmware handle this? If AMO, yes.  return to U-mode
5) firmware cannot handle it, mret to S-mode tvec
7) Linux looks at the instruction
8) Linux handles the illegal instruction
9) sret to user program

There's nothing erroneous here.

The scenario I'm talking about is much simpler, no Linux.
1) No delegation is set
2) illegal instruction happens in U-mode
3) trap to M-mode
4) M-mode inspects, can I handle this? If AMO, yes.  return to U-mode
5) Else, signal error

Right now, 5 hangs instead of doing anything. I'm simply suggesting to send an error packet and then hang, instead.

> The assert should fire in cases where it is sufficiently unusual to
> indicate that something may be going wrong ( I think this was the original
> intent of the github issue?)

Neither of these scenarios has anything unusual going on in the RTL.

farzamgl, 2020-01-21:
I think the way it happens in the firmware is the second scenario(no delegation).
If the firmware doesn't know how to handle and illegal instruction exception, can't we just send a finish packet to the host with the error code?

dpetrisko, 2020-01-21:
Yep, that’s what I’m suggesting if that was unclear. Do you know how Linux can handle illegal instruction signals? Like how is it able to intercept them? I would think it’s part of the SBI?

farzamgl, 2020-01-21:
They are handled in the firmware and if there's an unexpected instruction(or exception in general) the firmware prints an error line and calls a power-off function which right now just send the BP finish packet.

taylor-bsg, 2020-01-21:
>    6. Linux looks at the instruction
>    7. Linux handles the illegal instruction
>    8. sret to user program

Yep I am saying that an RTL assertion that emits a warning at step 5 (when
our firmware does not silently handle it, an mret back to supervisor) would
be handy.

Regarding Linux, and what Farzam said, I do not think it makes sense for
the OS to halt if a user program sees an illegal instruction.

dpetrisko, 2020-01-21:
What would the condition be for the assertion?

I agree on the OS side. I think the case is that current RISC-V hardware does not delegate illegal instruction exceptions, so the Linux support has never been added. There’s no reason not to patch that into Linux, but that’s the state of art

taylor-bsg, 2020-01-21:
Maybe that's only true in OpenSBI? I can't imagine that SiFive would allow
a Rocket core to be hung by a user instruction with an illegal instruction.

Regarding the condition, basically m-mode handoff of an illegal instruction
exception (any exception?) to a lower level of privilege..

dpetrisko, 2020-01-21:
I also imagine the SiFive Firmware has a mechanism to communicate error that’s not hanging. 

But M-mode handoff is not an error case. The error case is if M-mode is unable to handoff. Or is this assertion you’re proposing merely informative?

dpetrisko, 2020-01-21:
Basically, my position is that RTL assertions are useful for hardware error conditions and this is fundamentally a software concern. 

Coupling an RTL assertion to a certain sequence of instructions is just...wrong. If the firmware changes or the hardware changes we’ll need to modify the assertion. Especially when there’s no hardware overhead in sending a packet on error-full termination. It also has the advantage of working on ASIC/FPGA.

taylor-bsg, 2020-01-21:
> I also imagine the SiFive Firmware has a mechanism to communicate error
> that’s not hanging.
>
> But M-mode handoff is not an error case. The error case is if M-mode is
> unable to handoff. Or is this assertion you’re proposing merely informative?
Gathering together snippets from my previous posts (I realize it was spread
across several posts, so it may not be clear):

- The assert should fire in cases where it is sufficiently unusual to
indicate that something may be going wrong
- Yep I am saying that an RTL assertion that emits a warning

taylor-bsg, 2020-01-21:
> Basically, my position is that RTL assertions are useful for hardware
> error conditions and this is fundamentally a software concern.
>
> Coupling an RTL assertion to a certain sequence of instructions [...]
My experience is software people really like RTL assertions that tell them
unusual things are happening, because they get tired of staring at the
terminal for hours wondering if the program is making progress or if it is
hung and they are just wasting their time. For example, printing out when a
tile is unfrozen is a hit feature with HammerBlade.  Sure, if the software
stack does it a million times, and then you want to turn it off.

The memory mapped location is nice, but it is very brittle. Would it work
in LiteX sim? Or somebody grabs single core BlackParrot and sticks it in
their system?