black-parrot/black-parrot #300: SV vs Verilog files

alainmarcel, 2020-01-22:
On SV-Tests, I have the following error (In Surelog) on Blackparrot, "logic" is a SystemVerilog construct, you either need to make your file .sv or put a -sv prior to the file or -sverilog on the overall command line to treat all the files as SysteVerilog files. Right now, it is a .v file which should follow the Verilog syntax (which I flag).
SYNTX:PA0207] third_party/cores/blackparrot/bp_be/src/v/bp_be_calculator/bp_be_int_alu.v:58 Syntax error: no viable alternative at input 'logic signed',logic signed [reg_data_width_lp-1:0] src1_sgn , src2_sgn; ^-- ./slpp_all/work/third_party/cores/blackparrot/bp_be/src/v/bp_be_calculator/bp_be_int_alu.v:59 col:6.[WARNI:PA0205] third_party/cores/blackparrot/external/basejump_stl/bsg_noc/bsg_noc_pkg.v:4 No timescale set for "bsg_noc_pkg".

taylor-bsg, 2020-01-22:
Thanks Alain. Maybe should be a bug filed against sv-tests, that it is not
invoking surelog correctly on BlackParrot with the -sverilog flag?

alainmarcel, 2020-01-23:
I'll try on the sv-tests front if it can be fixed there, but ideally, sv-tests should be invoking the BlackParrot Makefile and all flags should come out correctly to match one of the main simulators.

dpetrisko, 2020-01-23:
Hi, could you point to where in sv-tests the command line parameters are getting set for SureLog?

For example, for VCS we add this flag in our makefile:
https://github.com/black-parrot/pre-alpha-release/blob/master/bp_common/syn/Makefile.vcs#L15

dpetrisko, 2020-12-02:
Files are now systemverilog in dev

PR #688 (2020-11-22): Changing files to standard systemverilog extension
Having SystemVerilog file extensions with .v causes problems with countless tools, requiring workarounds. Since SystemVerilog-2005+ is a requirement for BlackParrot, it is more clear that the file extensions follow this standard.

Fixes #300
closed 2020-12-02 by dpetrisko