black-parrot/black-parrot #301: Investigate Verible

dpetrisko, 2020-01-22:
https://github.com/google/verible

Verible has a code linter and formatter, conforming to the LowRISC style guide. Since the LowRISC style guide is fairly similar to our own, we should see if it’s possible to tweak the differences out (or modify our style guide slightly). Autoformatting would be a great feature.

taylor-bsg, 2020-01-22:
Linting is great.

Style — Tim Ansel pointed out that this is a nice way to not annoy part
time contributors with style policing of pull requests — just auto run it
on new code.

For existing modules, it is a little problematic in terms of gamification
and also tracing git history for actual substantive changes. Tim and I
discussed this problem a little bit, but the solution seemed complicated. I
wonder if we could have it reformat only changed lines? :-)

Seems like a nice project for interested external contributors to get this
to match the pre-existing BSG style guide.

M

dpetrisko, 2020-01-22:
> Style — Tim Ansel pointed out that this is a nice way to not annoy part
time contributors with style policing of pull requests — just auto run it
on new code.

Yep! That’s the biggest benefit. 

> git history for actual substantive changes

I’ve found that as long as feature commits are separated from style commits, this doesn’t end up being too painful. 

> gamification

Definitely something to think about. There are a few schools of thought on this. 

> Seems like a nice project for interested external contributors to get this
to match the pre-existing BSG style guide.

Agreed!

omasanori, 2022-02-21:
I would tackle this.

> For existing modules, it is a little problematic in terms of gamification and also tracing git history for actual substantive changes. Tim and I discussed this problem a little bit, but the solution seemed complicated. I wonder if we could have it reformat only changed lines? :-)

Yes, Verible seems to have a [Git integration](https://github.com/chipsalliance/verible/tree/master/verilog/tools/formatter#incremental-formatting) script for that purpose.

dpetrisko, 2022-02-21:
Hi, that would be an extremely useful tool to have set up, with the incremental formatter. Please keep us updated on the progress!

omasanori, 2022-02-22:
I found that [Bazel](https://bazel.build/) is required to build Verible. May I add it to the prerequisites (like Make, CMake, etc.)?

dpetrisko, 2022-02-22:
Sure, note that in the PR. We'll have to add it several places, including the Ubuntu docker image

omasanori, 2022-02-22:
> Sure, note that in the PR. We'll have to add it several places, including the Ubuntu docker image

OK, I will note in the description and update `docs/eval_guide.md` in the PR.

Note: https://github.com/black-parrot/black-parrot-sim and https://github.com/black-parrot-sdk/black-parrot-sdk will need an update at least.

omasanori, 2022-02-22:
After preprocessed by Surelog:
```
$ verible-verilog-lint bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:5:15: Explicitly define a storage type for every parameter and localparam, (width_p). [Style: constants] [explicit-parameter-storage-type]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:6:17: Explicitly define a storage type for every parameter and localparam, (els_p). [Style: constants] [explicit-parameter-storage-type]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:7:17: Explicitly define a storage type for every parameter and localparam, (ready_THEN_valid_p). [Style: constants] [explicit-parameter-storage-type]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:9:18: Explicitly define a storage type for every parameter and localparam, (ptr_width_lp). [Style: constants] [explicit-parameter-storage-type]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:107:40: Remove trailing spaces. [Style: trailing-spaces] [no-trailing-spaces]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:108:38: Remove trailing spaces. [Style: trailing-spaces] [no-trailing-spaces]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:109:93: Remove trailing spaces. [Style: trailing-spaces] [no-trailing-spaces]
bp_common/src/v/bsg_fifo_1r1w_rolly.sv.pp:110:37: Remove trailing spaces. [Style: trailing-spaces] [no-trailing-spaces]
```
`Style: trailing-spaces` messages are from preprocessed `BSG_ABSTRACT_MODULE`. Pseudo-preprocessing `BSG_INV_PARAM` would work well.

dpetrisko, 2022-02-22:
Seems like it might be difficult if we have to preprocess with Surelog before Verible flagging linter issues. Since then Verible would be reformatting the processed code rather than the original. Or did you have another idea for a flow?

omasanori, 2022-02-22:
For linting, I think that simply replacing `BSG_INV_PARAM(param)` with `param`  will work as expected. I will avoid to use Surelog for this purpose as the true preprocessing may modify unwanted lines, which is why I said *psaudo-preprocessing.*

Formatting is not trivial as you mentioned. Essentially, we need to do:

- List up changed lines, (Git does this part)
- Strip `BSG_INV_PARAM`
- Format only the changed lines in the first step (Verible has the `--lines` option to format only specific lines)
- Restore `BSG_INV_PARAM` (even if parameters moved to another lines after formatting!)

The last step will need a clever trick. I have rough ideas, but implementing only linting part first may be reasonable.

omasanori, 2022-02-25:
Hm, `declare_bp_proc_params` was another source of syntax errors. Perhaps preprocessing with Surelog is needed, as @dpetrisko pointed out, and then formatting is not feasible, at least until Verible's parser becomes more robust.

omasanori, 2022-02-25:
After some experiments, integrating Verible seems indeed tricky, and ad-hoc solutions are fragile. The good news is, however, Surelog and Verible are surprisingly active projects. There are people interested in this field like https://github.com/chipsalliance/Surelog/issues/1712#issuecomment-1046544390, so I am confident that the situation will be getting better.

dpetrisko, 2022-02-25:
Thank you for checking this out!