black-parrot/black-parrot #323: Make the fe_queue / fe_cmd fifos optionally bypassable

dpetrisko, 2020-02-19:
One easy tradeoff between critical path and performance is bypassing fifos if empty.  This may include modifications to the basejump_stl fifo implementations as well (or a new fifo type).  Critical path + performance analysis should be done to make sure this design point is pareto optimal

taylor-bsg, 2020-02-19:
Interesting. Bypass and FIFO could mean three things:

a. having a mux that muxes between the FIFO input and the FIFO contents,
basically allowing a zero latency path, but with some cycle time impact.

b. having a mux that holds the head element and the value of valid in a
flip flop, so there is only a CLK->Q latency on the FIFO. this still has a
latency of one cycle but short cycle time. The main catch is that the
critical path on the yumi (then_ready) signal is longer, since it is a late
signal that will control the data path mux rather than just a mux on a read
ptr.

For  fifos implemented out of shift registers, B can be implemented just by
moving the muxes before the flops, and little area impact is anticipated.
For RF based FIFOs, there will be an extra register of overhead.

c. a combination of A and B, where the data and valid element can be
bypassed but otherwise comes from a flip flop.

dpetrisko, 2020-03-19:
Good ideas! I had only pictured a) when I wrote the issue.

 For the fe_queue, we can't use a bypass flip flop because it still needs to be stored in the rolly fifo in case of rollback.  From my sketching, it seems very difficult to bypass the fifo while maintaining the rolly properties.

For the fe_cmd fifo, any of those could work. The path is tight on both ends (branch compare->fifo, fifo->pc->I$), so b might make the most sense

dpetrisko, 2020-04-01:
#390

dpetrisko, 2022-06-27:
This turns out to be worse for critical path than the performance gain

closed 2022-06-27 by dpetrisko