black-parrot/black-parrot #327: Handshaking refactor

dpetrisko, 2020-02-20:
Currently, we have a clarity problem where there are no differences between rv& and r->v as indicated by the interface names.  This has caused several bugs where rv& was assumed and connected to a r->v module.

Proposal:
```
rv&:
  ready_x
  v_x
r->v:
  ready_x
  then_v_x
v->r:
  v_x
  yumi_x
```

Then the taxonomy is a little more clear: [paper](http://cseweb.ucsd.edu/~mbtaylor/papers/Taylor_DAC_BaseJump_STL_2018.pdf)

```
demanding consumer (v->r) / helpful producer (v->r):
FIFO

helpful consumer (rv&) / helpful producer (v->r):
ready_o(ready_lo)
v_i(v_li)

v_o(v_li)
yumi_i(ready_lo & v_li)

helpful consumer (rv&) / demanding producer (r->v):
ready_o(ready_li)
v_i(ready_li & then_v_lo)

ready_i(ready_li)
then_v_o(then_v_lo)

demanding consumer (v->r) / helpful producer (v->r):
v_i(v_li)
yumi_o(yumi_lo) 

v_o(v_li)
yumi_i(yumi_lo)
```

I suggest keeping yumi for now, just so that there's less churn in the code base

@taylor-bsg 
@stdavids

stdavids, 2020-03-18:
I think the rules for connecting handshakes has a bug. If I have the following interfaces: output of a module is `ready_i -> then_v_o` and input to another module is `v_i -> then_ready_o` then I cannot connect these two directly because both are demanding (I must use a fifo or something). However in your connection rules, rule 1 states that `ready_i` can accept `then_ready_o` which in this case is not true. It is only true if the output interface was `ready_i & v_i`. I don't think there will be a simple 1:1 relationship there because you use the same names `ready_x` and `v_x` in multiple interfaces. I think you need to uniqify all of the names to be able to establish such a rule.

With that said, I like the names so I propose we stick with the helpful/demanding paradigm and have the following:

1. `ready_x` & `v_x` is always helpful (both as a data input and data output handshake)
2. `v_i` -> `then_ready_o` is a data input handshake and is demanding
3. `ready_o` -> `then_v_i` is a data input handshake and is helpful
4. `v_o` -> `then_ready_i` is a data output handshake and is helpful
5. `ready_i` -> `then_v_o` is a data output handshake and is demanding

Another way to think about this is that an interface is helpful if the output control signal can/must come first. A demanding interface requires the input control signal before it will assert the output control signal.

So then connection rules are simply that the only illegal combo is two demanding interfaces.... However, we get this case with number 3 and 4 where it is helpful but an order is still mandated which leads to cases where we sometimes need an & gate when connecting with another helpful interface to make sure signals don't go early... which muddies the water a bit.

So here is my thought, what if we never used 3 or 4? I don't believe we ever have to if we add an & gate inside the module (I might be wrong...). Basically, we push that & gate inside the module where we either:
1. needed it and had to create it outside the module
2. The signal was going to come after so logically it is no different we just have a redundant & gate

My opinion is that and gate is negligible given how much it simplifies this paradigm as now we just have:

1. `ready_x` & `v_x` is always helpful (both as a data input and data output handshake)
2. `v_i` -> `then_ready_o` is a data input handshake and is demanding
5. `ready_i` -> `then_v_o` is a data output handshake and is demanding

Now the rule is truly that demanding cannot connect to demanding. No additional & gates ever needed for direct connections. And further more, you can tell an interface is demanding because one of the ports has "then_".

Thoughts? I'd like to hear from @taylor-bsg too because I know this is purposefully adding hardware to a lot of places but I think even on the scale of a full chip you wouldn't notice.

dpetrisko, 2020-03-18:
> think the rules for connecting handshakes has a bug. If I have the following interfaces: output of a module is ready_i -> then_v_o...

Yeah, you're right there.  Like you say, the rule is actually "then_* can only connect to rv&" (have to look at both ready and valid of the other module).

dpetrisko, 2020-03-18:
> So here is my thought, what if we never used 3 or 4?

100% agree. I actually meant to remove those from the proposal when I went through the taxonomy chart and noticed I never used them :)

stdavids, 2020-03-18:
4 is used quite a lot in basejump_stl, this is the classic v_o, yumi_i that is found in fifos and many other modules, but I don't think i've ever seen a case where that can't be changed by pushing an & gate inside the module and then it becomes a ready & valid interface.

stdavids, 2020-03-29:
How do we feel about reducing ready to just `r` like we do with valid? The major conflict is that we use `*_r` for register outputs (ie. `foo_r`) and this can pop up into our port list when an output port is directly connected to a registers (ie. `foo_r_o`). This might be more of a basejump stl thing, where we create modules that register inputs quickly and we tend to let the output have combinational delay, so when an output port is directly registered we will use the `*_r_o` but maybe we can change that to be `*_ro`?

dpetrisko, 2020-03-29:
r also conflicts with ‘read’

dpetrisko, 2020-03-29:
I do like the _ro change, though

stdavids, 2020-03-29:
What, you don't like `r_r_o`? What could be more clear?! Hahaha, I agree though, I think keeping ready is probably best. v gets a lucky pass that very few things use v.

taylor-bsg, 2020-07-02:
Taking your thoughts and suggestions in place, here is what I propose for consistency and easy use (it turns out we were already using ready_and and ready_then in bsg_noc_links, as suggested I am resting yumi)

valid, ready_and
valid, ready_then
valid, yumi
valid, credit

Benefits:

1. Incremental change to existing code. Just throw the right suffix on the ready signal.

2. Determining errors in the handling of a handshake comes down to the following (usually localized) checks:

a. interface mismatch (simply see if the ready signal name is different)

b. not putting the “and gate” on the ready_and.

c. using ready_and or yumi in the input logic cone of the valid signal 

d. using valid signal to derive ready_then 

e. using anything but credit or ready_and on a top-level interface

taylor-bsg, 2020-07-04:
I took a pass at adding to the SystemVerilog style guide, and significantly refined it. This may be the best description yet. Take a look!

https://docs.google.com/document/d/1xA5XUzBtz_D6aSyIBQUwFk_kSUdckrfxa2uzGjMgmCU/edit

dpetrisko, 2020-07-04:
Looks great!

muwyse, 2020-12-04:
@dpetrisko @taylor-bsg I think it is time we make a pass at our code base and implement this code clarity refactor. I just ran into this again while integrating the new FSM CCE that uses the BP Burst interface to remove serialization/deserialization. Turns out the burst2lite converter uses r&v while bp_cfg uses r->v and these two do not mix well when the producer is r&v and consumer is r->v.

taylor-bsg, 2020-12-04:
Hiya Mark,

I suggest code refactors going forward go ahead and update the names
according to what I put in the SV guide...

M

dpetrisko, 2020-12-04:
Yep, we should enforce the style guide “ready_and_i, ready_then_i, etc. ” for all new modules. Old modules should be cleaned whenever they’re touched. 

A bulk refactor is also fine, but it’s not purely style. Some thought needs to go into making modules ready-and vs ready-then