black-parrot/black-parrot #329: Unnecessary message types in memory interface

dpetrisko, 2020-02-21:
The memory interface should only define read and write commands.  Information about uncached / prefetch is only relevant as part of the return type and therefore should be part of the payload, not the main message.

https://github.com/black-parrot/pre-alpha-release/blob/master/bp_me/src/include/v/bp_me_cce_mem_if.vh#L22

Another advantage is that lets peripherals be a little bit leaner by defining smaller payloads

muwyse, 2020-02-21:
I'm not convinced we want it to be that restrictive. At a minimum I think we want read, write, prefetch, and an uncached bit to be visible to the memory-side controllers (i.e., not hidden in the payload). For example, if the CCE connects to an L2, the L2 may be able to read only a subset of a cache block (saving power, and maybe cycles) for an uncached request (this also depends on what we mean by uncached; our current design has the L2 being a victim cache on the memory-side, so we can imagine an uncached access from a CPU being served from the L2). Additionally, a prefetch could be ignored by the L2 (with the L2 simply responding immediately to the CCE and not prefetching the block) if, for example, the current load on the L2 or memory controller is high, where adding additional commands may further congest the networks or delay non-prefetch (required) accesses.

Further, it is likely (from a research standpoint) that we may want to support custom message types / commands that are visible to the memory controllers (i.e., not hidden in the payload). The memory controller could then issue simple read/write style commands, but it might be useful for it to accept more complicated commands and then implement their functionality.

dpetrisko, 2020-02-21:
You've convinced me on the prefetch bit, but I think the uncached bit is confusing as currently implemented.  It currently indicates uncached in L1, but it is sent to L2 / I/O devices.  It's caused bugs where I/O devices send back uncached responses to cached requests (when it should be perfectly valid to cache I/O in some cases).  

I think maybe what would solve my confusion is to have the uncached-in-L1 bit live in the payload, but the uncached-in-L2 bit live in the message type.

>  the L2 may be able to read only a subset of a cache block (saving power, and maybe cycles) for an uncached request
I think this is solved by the size field

> Further, it is likely (from a research standpoint) that we may want to support custom message types / commands that are visible to the memory controllers (i.e., not hidden in the payload).
There's no reason a "researchy" device couldn't inspect the payload. I think it makes sense to hide it by default unless supported.

Nit: do we need the speculative bit in the payload, if it's a prefetch request shouldn't it return a prefetch response?

dpetrisko, 2020-02-21:
I think maybe what I'm proposing here is to differentiate between features needed to adhere to this interface and features which are simply used for optimization.

In my mind, a device need only :
1) implement reads and writes, uncached (in L2) reads and uncached (in L2) 
2) Forward back all other fields 

In this way, all of the message metadata is in the payload (uncached in L1, speculative, etc.).  Whether to inspect it is up to the device receiving the message.

Otherwise, we incur a code burden of needing to implement checks for message types which we don't support in a device (uncached in bsg_cache, etc.) and also need to add new behavior to adapters / clients / etc. every time we add a new message which may be used only in a single device (especially researchy messages)

muwyse, 2020-04-10:
I think we are converging. L2 (or really, memory beyond CCE) operations should be:
- Load / read
- Store / write
- Uncached load / uncached read
- Uncached store / uncached write
- Prefetch block

* uncached means uncached in L2 or memory level beyond CCE

All requests have a size field, which can be 1 to 64+ bytes by power of 2. Address must be aligned to request size. Messages also have a payload field that is passed blindly between hierarchy levels and used only by the CCE. Payload will include things like what LCE requested the operation, if the access was speculative by the CCE, if the access was cached/uncached by the LCE, possibly a custom message field for use by CCE or research devices, etc.

muwyse, 2020-04-25:
closed by merge #453

closed 2020-04-25 by muwyse