black-parrot/black-parrot #330: Add nonsynth deadlock detector

dpetrisko, 2020-02-21:
Often, tests fail by hanging because of X-propagation or bad instructions.  They are then left to wait for timeouts, which are set high for long-running tests.  We should implement a halt detector which checks if PC has changed in N cycles, where N is ~100000.  Upon detecting a halt, it should loudly exit the simulation.

taylor-bsg, 2020-02-21:
Interesting! I guess for gate-level, you have fewer options, but for RTL,
we could also have an X detector module in bsg_tests to more directly
detect X without any magic number? For gate-level, you can reach down and
find the registers but it requires scripting that can be a little brittle
(although we do have pyverilog that we use for sv2v).

M

dpetrisko, 2020-07-11:
Watchdog timer / PC==x detector was added a ways back

closed 2020-07-11 by dpetrisko