black-parrot/black-parrot #340: Getting started: missing prerequisites & command execution order problem

andreaskuster, 2020-02-25:
[`https://github.com/black-parrot/black-parrot/blob/master/GETTING_STARTED.md#getting-started-full`](https://github.com/black-parrot/black-parrot/blob/master/GETTING_STARTED.md#getting-started-full)

I followed the guide `Getting started (Full)` and ran into the following issues (`Ubuntu 19.10, x86_64, latest version`):

### 1.  `make tools`: two packages are missing, namely: `device-tree-compiler` and `byacc` (see error log below).  Please add them to the prequisite apt-get line for ubuntu.

```
...
make[1]: Entering directory '/home/osboxes/pre-alpha-release'
Submodule 'external/riscv-isa-sim' (https://github.com/black-parrot/riscv-isa-sim.git) registered for path 'external/riscv-isa-sim'
Cloning into '/home/osboxes/pre-alpha-release/external/riscv-isa-sim'...
Submodule path 'external/riscv-isa-sim': checked out '4f664cf76926aae83f64b8cf4722aba392147762'
cd /home/osboxes/pre-alpha-release/external/riscv-isa-sim; \
	./configure --enable-commitlog --enable-histogram --with-isa=rv64ia --prefix=/home/osboxes/pre-alpha-release/external \
		--with-fesvr=/home/osboxes/pre-alpha-release/external
configure: WARNING: unrecognized options: --with-fesvr
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for ar... ar
checking for ranlib... ranlib
checking for dtc... no
configure: error: device-tree-compiler not found
make[1]: *** [/home/osboxes/pre-alpha-release/external/Makefile.tools:44: spike] Error 1
make[1]: Leaving directory '/home/osboxes/pre-alpha-release'
make: *** [Makefile:37: tools] Error 2
```
```
...
cp /home/osboxes/pre-alpha-release/external/axe/axe /home/osboxes/pre-alpha-release/external/bin
make[1]: Leaving directory '/home/osboxes/pre-alpha-release'
make cmurphi
make[1]: Entering directory '/home/osboxes/pre-alpha-release'
Submodule 'external/cmurphi' (https://github.com/black-parrot/cmurphi.git) registered for path 'external/cmurphi'
Cloning into '/home/osboxes/pre-alpha-release/external/cmurphi'...
Submodule path 'external/cmurphi': checked out '6b28ad932f0e6d10e4a78be8ca57e4b3f2476f5d'
make -C /home/osboxes/pre-alpha-release/external/cmurphi/src mu 
make[2]: Entering directory '/home/osboxes/pre-alpha-release/external/cmurphi/src'
byacc -dt mu.y
make[2]: byacc: Command not found
make[2]: *** [Makefile:128: y.tab.h] Error 127
make[2]: Leaving directory '/home/osboxes/pre-alpha-release/external/cmurphi/src'
make[1]: *** [/home/osboxes/pre-alpha-release/external/Makefile.tools:84: cmurphi] Error 2
make[1]: Leaving directory '/home/osboxes/pre-alpha-release'
make: *** [Makefile:39: tools] Error 2
```

### 2.  `make ucode`: Failed with the error message that `bsg_ascii_to_rom.py` was missing (see error log below). I was able to resolve this problem by running `make libs`  before `make ucode`. The reason is that `make libs` seems to fetch the missing files i.e. `git submodule update --init --recursive $(BP_EXTERNAL_DIR)/basejump_stl` (see makefile below).

```
...
(74) bi ready
(75) error: stall
rm /home/osboxes/black-parrot/bp_me/src/asm/microcode/cce/fast.pre
make[2]: Leaving directory '/home/osboxes/black-parrot/bp_me/src/asm'
python2: can't open file '/home/osboxes/black-parrot/external/basejump_stl/bsg_mem/bsg_ascii_to_rom.py': [Errno 2] No such file or directory
make[1]: Leaving directory '/home/osboxes/black-parrot/bp_me/src/asm'
```

```
## Find top of git repo
TOP ?= $(shell git rev-parse --show-toplevel)

include $(TOP)/Makefile.common
include $(BP_EXTERNAL_DIR)/Makefile.tools

.NOTPARALLEL:
.PHONY: update_submodules update_tests tools progs ucode
.DEFAULT: update_submodules

## This target updates submodules needed for building BlackParrot.
#  We only need to keep update basejump_stl up to date. The other submodules
#    are for building tools, which we should only need to do every so often

update_libs:
	cd $(TOP); git submodule update --init --recursive $(BP_EXTERNAL_DIR)/basejump_stl

update_tests:
	cd $(TOP); git submodule update --init --recursive $(BP_COMMON_DIR)/test

tidy_tools:
	cd $(TOP); git submodule deinit -f external/

bleach_all:
	cd $(TOP); git clean -fdx; git submodule deinit -f .

## This target fetches and builds all dependencies needed for running simulations
#    to test BlackParrot. By default, all tools are built but comment any tools that 
#    you already have a copy of. If your version of a tool significantly differs from 
#    our submodule version, use at your own risk.
#
tools:
	$(MAKE) gnu
	$(MAKE) systemc
	$(MAKE) verilator
	$(MAKE) fesvr
	$(MAKE) spike
	$(MAKE) dromajo
	$(MAKE) axe
	$(MAKE) dramsim2
	$(MAKE) cmurphi

progs: update_tests
	$(MAKE) -C $(BP_COMMON_DIR)/test all_mem all_dump all_nbf

ucode:
	$(MAKE) -C $(BP_ME_DIR)/src/asm roms

libs: update_libs
```

### 3. This is not an actual error, since the redirection seems to work fine, but I suggest to adjust the repo url from `git clone https://github.com/black-parrot/pre-alpha-release.git` to the actual url `https://github.com/black-parrot/black-parrot.git` (in quickstart and full).

dpetrisko, 2020-02-25:
Hi, thanks for the issue! We're aware of the instruction issues. They are currently updated in the branch 'uw_ee477_pparrot_wi20' (where we're test-piloting our code).  Once we flush out bugs there, the fixes will all propagate to master

The Ubuntu dependencies are good to know!

taylor-bsg, 2020-02-25:
Hi Andreas, thanks for posting the issue! Sorry for the confusion, the instructions to follow are here:

https://github.com/black-parrot/black-parrot/blob/uw_ee477_pparrot_wi20/README.md

Be sure to  perform the 

git checkout uw_ee477_pparrot_wi20

command to checkout the class branch, per the instructions.

dpetrisko, 2020-03-10:
Closing this, documentation is updated in master

closed 2020-03-10 by dpetrisko