black-parrot/black-parrot #343: Add more information to Interface Verifier

dpetrisko, 2020-02-26:
bp_nonsynth_if_verif currently prints out the top level parameters and a handful of interface widths. It would be helpful to see some more derived statistics such as:

1) Message header/data field sizes
2) Flits per message type / flits per header
3) Total cache size
4) ... etc.

dpetrisko, 2026-03-09:
Partially fixed with https://github.com/black-parrot/black-parrot/pull/1241, leaving open for further improvements!

Mayuri-pawar21, 2026-03-24:
Hi @dpetrisko  , I've opened pull request #1272  to add L2 cache size, total cache size, and number of cores as further enhancements for this issue. Please review when you have time.

dpetrisko, 2026-09-08:
Closing due to vagueness in definition of "done" -- improvements are always welcome :)

closed 2026-09-08 by dpetrisko