black-parrot/black-parrot #360: Combine LCE cmd msg / uncached msg

dpetrisko, 2020-03-03:
We can just reuse the uncached_only mode of the normal LCE cmd module.  Having the second module is extra code and introduces weird intermediate states during the transition.

dpetrisko, 2020-03-08:
Another design point is to have both an I/O CCE and regular CCE in the tile. Advantage is code deduplication. Disadvantage is that CCE ucode wouldn’t be able to inspect I/O messages. That’s probably enough of a downside to kill this idea

muwyse, 2020-03-11:
These two modules are being combined in my CCE v3 changes, which should help with the transition between uncached only and microcode operation.

muwyse, 2020-04-30:
Closing; design no longer has separate modules.

closed 2020-04-30 by muwyse