black-parrot/black-parrot #366: Move testbenches to pure C++

dpetrisko, 2020-03-06:
We don't actually use any SystemC features (confirm @muwyse?), instead just wrapping our testbenches with a thin clock/reset driver.  But we're stuck with the overhead of compiling in systemc libraries. Small rewrites could easily port these testbenches to C++.

muwyse, 2020-03-11:
Is there a path where we can also support SystemC testbenches? Perhaps we change the current .sc target to .vlt (or whatever the correct verilator abbreviation is), and then .sc would be verilator+systemc?

Two of the (likely currently broken) ME tests for individual CCE components did use SystemC. I can see some possible usefulness for having SystemC supported in our build system, but not necessarily for our main regressions that only feed in clock+reset.

dpetrisko, 2020-03-11:
Oh, interesting! This definitely isn’t urgent. Could you point me to the systemc features being used?

muwyse, 2020-03-12:
Not sure I'm using any advanced SystemC features or doing anything that can't be done in a pure C++ testbench, but bp_me/test/tb/bp_cce_alu/test_bp.cpp uses SystemC. The SystemC stuff drives the test instead of a verilog testbench. I personally think this approach may be easier than writing up tests in verilog/systemverilog or even using the bsg trace replay stuff because if you are decently good at C++ you will have no problem with SystemC. I know plenty of people dislike C/C++, but I honestly don't have issues using it over anything else, especially verilog.

I used to have a more advanced one that was for the whole CCE. This was more complex because you have to get the r&v/r->v handshaking correct in the SystemC code as well as deal with translating between bit vectors and C++ structs or data types for input/output ports. Although it is possible to develop a library that could help here.

dpetrisko, 2020-03-12:
Oh interesting. Yeah, I’m totally fine with having non-Verilog driven testbenches. Bsg_trace_replay is great for testbenches that need to be 1) synthesizable and 2) latency insensitive. But for more direct testing it can be cumbersome. 

Another thing we were looking into for basejump was cocotb (python driven testbench), although the Verilator support isn’t quite there yet. 

I don’t see any systemc-specific stuff.  Most likely, people are less familiar with systemc than c++, (plus Verilator seems to be leaning into the c++ support more, nowadays), so it probably is still good to rewrite so we have one less dependency / unknown. Like I say, not urgent but on the back burner.

developandplay, 2021-04-06:
With https://github.com/black-parrot/black-parrot/issues/798 the testbenches shouldn't depend on SystemC anymore.
So if there are no other use cases we could probably remove the dependency.

muwyse, 2021-04-12:
I don't have any current use-cases. I think this dependency is safe to remove.

dpetrisko, 2021-04-12:
closed with #834

PR #834 (2021-04-12): Remove systemc submodule - no longer used
Removes systemc submodule since it is no longer used, and removes one deprecated header file in ME related to old systemc testbench.

Closes #366
closed 2021-04-12 by dpetrisko