black-parrot/black-parrot #378: Investigate RV64E as BedRock base ISA

dpetrisko, 2020-03-11:
This would let us reuse the RISCV assembler (not compiler out-of-box, because we'd still only have 8 registers).  It would also reduce the barrier of entry for new BedRock programmers.  Since we're not using compressed instructions, a simple encoding for BedRock extensions would be:

[<31 bits> | 0] -> BedRock extension
[<31 bits> | 1]  -> RV64E

The CCE may also support only a subset of instructions, depending on PPA. (i.e. probably not supporting subword instructions, although I don't know what the current ISA looks like)

muwyse, 2020-03-11:
I will investigate. It certainly would be great to reuse the RISCV assembler and not require a custom one.

The microcode encoding is fairly tight for a couple of the instruction types (e.g., push/send message and branch conditional based on flag registers).

muwyse, 2021-08-17:
Current status: RV64E is not a defined ISA within RISC-V. I have heard some discussion within the community expressing desire for it to exist, but it does not seem to have a strong enough backing to actually make it happen. There may be a way to "force" the compiler to generate RV64I code that only uses the registers 0-15 with compiler flags, but I am not sure that those flags are guaranteed to avoid accessing registers 16-31. Existing compilers would also need to be coerced to not access memory when generating assembly from C code.

An alternative approach would be programming in RV64I assembly without using registers 16-31, adding custom instructions to the assembler for coherence operations (e.g., read directory, process directory), and then use the assembler to transform the RV64E-like assembly into machine code executable by the coherence engine.

dpetrisko, 2022-06-28:
Nice find!