black-parrot/black-parrot #382: Analyse design point of coherence D$, incoherent I$

dpetrisko, 2020-03-12:
Since we have both a coherent and non-coherent cache controller, this is a naturally interesting design point.  For normal operation, the changes we need to make are:

- configuring the LCE ids / coordinates etc to handle 1 LCE 
- parameterizing l1i separately from l1d
- parameterizing whether the cache is connected to the LCE or the UCE

For I$ fence, this is where things get fun.
- Need to force an illegal instruction, trap
- Send IPI to each other processor to force writebacks of D$
- Flush local I$ (I think actually, flush all I$.  Normally this is handled by the OpenSBI firmware, but in that case, we would end up with an infinite recursion of remote fencei's)

Probably it makes sense to implement remote fencei as a simple memory-mapped access instead of an IPI, because that would prevent us from having to use a custom instruction.

taylor-bsg, 2020-03-12:
Yeah one of the students pointed out that the i- and d- caches appear to be
parameterized with the same parameters. Any ideas on how they can easily
decouple it?

M

dpetrisko, 2020-03-12:
Very carefully :)

There's no trick to it, one just needs to go through all of the lce_sets / lce_assoc and replace with icache_sets / icache_assoc / dcache_sets / dcache_assoc.

And then need an assertion that icache_sets/assoc == dcache_sets/assoc if they're coherent, since this is a restriction that the CCE currently has (although I think this restriction will go away with Mark's recent work).

muwyse, 2020-03-12:
Correct, this restriction will go away. Part of my changes for CCE v3 separate out the lce_sets, lce_assoc, etc. to become icache_xxx, dcache_xxx, acache_xxx so we can support varied cache organizations for icache, dcache, accelerator caches.