black-parrot/black-parrot #392: code clarity

taylor-bsg, 2020-03-16:
https://github.com/black-parrot/black-parrot/blob/8a394000d4e17078ed08ea2da626f9d5e926eb60/bp_be/src/v/bp_be_mem/bp_be_dcache/bp_be_dcache.v#L101

we use page_offset_width_p and bp_page_offset_width_gp in this file interchangeably, and they are both set to 12 in the common_aviary_defines and bp_common_pkg files. We should eliminate page_offset_width_p and just use bp_page_offset_width_gp.

dpetrisko, 2020-03-16:
Probably rv64_page_offset_width_gp, but yeah agree

dpetrisko, 2020-03-16:
Probably all of these global variables should be fixed in bp_common:
https://github.com/black-parrot/black-parrot/blob/master/bp_common/src/include/bp_common_aviary_defines.vh#L263

```
  , localparam dword_width_p       = 64                                                            \
  , localparam instr_width_p       = 32                                                            \
  , localparam csr_addr_width_p    = 12                                                            \
  , localparam reg_addr_width_p    = 5                                                             \
  , localparam page_offset_width_p = 12                                                            \
```

dpetrisko, 2020-11-04:
#658

closed 2020-11-04 by dpetrisko