black-parrot/black-parrot #393: Correctness

taylor-bsg, 2020-03-16:
https://github.com/black-parrot/black-parrot/blob/9b3cea5c453d307f893044be6cd5c09f907cab9e/bp_be/src/v/bp_be_mem/bp_be_mem_top.v#L476

Should just take low 12 bits of vaddr, rather than using these fields, since we are just using this for translation.

dpetrisko, 2020-03-16:
Depends on what parameters we want to fix.  I think it's reasonable to have a dcache which only accepts a 12 bit vindex and reconfigures sets, based on cacheline size internally to support this, so this change is probably fine.

dpetrisko, 2020-03-16:
There's also the question of allowing synonyms in the I$...

taylor-bsg, 2020-03-18:
My point is, this code doesn't need to consider the index or the offset, it is only used for VM, which has no relationship to the cache parameters. There is no benefit to writing the code this way. It can just blindly take the low 12 bits. In all other places except this one, it just uses a constant which evaluates to 12.  So defining it this way is just creating an unnecessary mismatch in width for no benefit.

dpetrisko, 2020-03-18:
Oh, I misinterpreted and thought we were talking about the D$ packet, not the mmu_cmd.  Agree

dpetrisko, 2020-06-08:
So based off a discussion with Andrew W, this is actually incorrect behavior. We need to pass the entire 64-bit effective address in the mem_cmd, so that 
1) we can check that bits 39-64 match bit 38 in the case of translation enabled 
2) passthrough bit 39 in the case of translation disabled 

https://groups.google.com/a/groups.riscv.org/d/msgid/isa-dev/CABXpatqk%2BGinxy%3DV_u7K5iYB6N_wYuLeNA5YZSEhPy0BEqU6dQ%40mail.gmail.com?utm_medium=email&utm_source=footer

taylor-bsg, 2020-06-08:
I think you have attached this comment to the wrong issue... this comment is saying that the page offset field for VM should be populated just by taking the low 12 bits instead of this expression.

M

dpetrisko, 2020-06-08:
No, this is the right issue. The mmu_cmd struct needs to change to be 64 bit address, not split into page offset and vtag as the issue title says

dpetrisko, 2020-08-06:
fixed by #543

closed 2020-08-06 by dpetrisko