black-parrot/black-parrot #400: Cosim mismatch when running random instruction test [riscv_jump_stress_test_0]

kabylkas, 2020-03-20:
I was running several tests that are randomly generated with [riscv-dv](https://github.com/google/riscv-dv) tool from Google. Several of them are failing. This is one of them.

Dromajo error:
```
[error] EMU PC 00000000800072b8, DUT PC 00000000800072b9 <<-- PC is off
[error] EMU INSN 011d6bb3, DUT INSN 011d6bb3
[error] EMU MSTATUS a000c0080, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
```

Here is the binary and generated assembly: [riscv_jump_stress_test_0.zip](https://github.com/black-parrot/black-parrot/files/4357647/riscv_jump_stress_test_0.zip)

Failing test stresses jump instructions. To generate this test I used default riscv-dv's configuration for this test:
```
 71 - test: riscv_jump_stress_test
 72   description: >
 73     Stress back-to-back jump instruction test
 74   iterations: 2
 75   gen_test: riscv_instr_base_test
 76   gen_opts: >
 77     +disable_compressed_instr=1
 78     +instr_cnt=5000
 79     +num_of_sub_program=5
 80     +directed_instr_1=riscv_jal_instr,20
 81   rtl_test: core_base_test
```

dpetrisko, 2020-03-20:
Thanks for the report! I’ll look into it

dpetrisko, 2020-03-20:
BTW, please feel free to make a PR with your riscv-dv flow. We do have a flow in place, but it is very primitive, more of a proof of concept than anything

kabylkas, 2020-03-21:
> BTW, please feel free to make a PR with your riscv-dv flow. We do have a flow in place, but it is very primitive, more of a proof of concept than anything

Okay 👍 its kind of dirty right now. I will make a PR once I clean it up.

PR #489 (2020-05-11): Fixing LSB of branch target to always be 0
RISC-V spec defines JALR to zero LSB to 0, to prevent misalignment

Fixes #400 #402
closed 2020-05-26 by dpetrisko