black-parrot/black-parrot #404: LCE req should use data length field

dpetrisko, 2020-03-23:
Uncached requests are assumed to be 64 bits max.  While we still need the uc_req_size field in order to perform subword uncached accesses, we also want to support up to cache line size uncached transfers, for host interfacing or DMA.

https://github.com/black-parrot/black-parrot/blob/top_dev/bp_top/src/v/bp_io_link_to_lce.v#L90

muwyse, 2020-03-24:
I will take a look at adding support for uncached cache block size load/store after the CCE v3 is tested and in (in the coming days). I think this will be relatively straightforward to support.

muwyse, 2020-04-25:
Merge #453 adds functionality to enable LCE Requests up to cache block size. More changes are needed to modify the LCE-CCE networks and endpoints (LCEs and CCEs) to fully handle this case. For example, the LCE Request network is currently instantiated system wide to only support up to 64-bits of data. This will need to change at the CCEs and the LCEs that want to send uncached stores up to cache block size.

It may make sense to roll this together with fixes for #260 to avoid unnecessarily wide connections. Or, we would want wormhole adapters that can be customized for the endpoint. For example, if an LCE will never do an uncached store larger than 64-bits, it doesn't need to connect to its wormhole adapter with an LCE Req struct setup to handle up to 512-bits.

dpetrisko, 2020-04-26:
> For example, the LCE Request network is currently instantiated system wide to only support up to 64-bits of data. This will need to change at the CCEs and the LCEs that want to send uncached stores up to cache block size.

Could you point to how this is handled currently?

> Or, we would want wormhole adapters that can be customized for the endpoint.

That's easy enough, just need to change the payload width to correspond to the max data size

https://github.com/black-parrot/basejump_stl/blob/c04be76f99eee6c1f1a75e6a22f6c5de34707132/bsg_noc/bsg_wormhole_router_adapter_in.v#L11

muwyse, 2020-04-27:
Here is my initial plan:
1. change "data_width_mp" param name in LCE-CCE IF macro to "lce_req_max_data_width_mp" to clarify naming. Add comments.

2. modify the encoder files in bp_me/src/v/wormhole to properly compute the wormhole length for all messages.
2a. Uncached LCE store request message size needs to be based on the size field of the message
2b. Uncached LCE Command data message size needs to be based on the size field of the message

3. fix instantiations/uses of the interface throughout the system. CCEs will support up to block width uncached accesses. D$ and I$ will support up to 64-bit uncached access. Other LCEs can be configured to support max uncached access size from 1B to block width bytes.

dpetrisko, 2020-04-27:
Sounds good!

dpetrisko, 2020-04-27:
This can be another issue, but this strategy could be applied for the lce_cmd/lce_resp as well. For example, accelerator caches only need to support acache_block_width_p for lce_cmd/lce_resp

dpetrisko, 2020-05-19:
Any update on this?

muwyse, 2020-05-19:
I have a branch started for this. Making the changes to the interface and the encoders is easy, and done I think. I also have changes for the CCE.

There is still work to do for linking together everything since we want to declare some of the LCE networks to be narrow, and some wide. This could be done as a second change though, after committing the interface and CCE changes.

muwyse, 2020-05-30:
Enabled by #523

closed 2020-05-30 by muwyse