black-parrot/black-parrot #405: Add a comment here to justify these sizes

taylor-bsg, 2020-03-24:
https://github.com/black-parrot/black-parrot/blob/8a394000d4e17078ed08ea2da626f9d5e926eb60/bp_common/src/include/bp_common_aviary_pkg.vh#L43

taylor-bsg, 2020-03-24:
,fe_queue_fifo_els: 8
      ,fe_cmd_fifo_els  : 4

I would assume that fe_cmd_fifo only needs to be 1 or 2 elements?

dpetrisko, 2020-03-24:
No actually. This caused a very subtle bug a ways back. We have two entry points for writing in the fe_cmd queue. We can write at EX1 for attaboy/pc_redirect, or we can write at EX3 for traps.  Then, we may have back to back writes, along with a latent attaboy inside the queue. Because we cannot stall, we must allocate enough entries so that the queue will always be ready by the time a trap rolls around. 

We should both comment and assert that’s its true. 

Aside: I think sending attaboys at EX3 would solve this and allow us a 2 entry fe_cmd fifo. And we may want to do it anyway for security / predictor accuracy reasons

taylor-bsg, 2020-03-24:
Agree with asserting and commenting this info.  Attaboys: with some benchmarks we will be able to see if there are any performance implications!

M

dpetrisko, 2021-02-25:
Formalized and fixed with #772

closed 2021-02-25 by dpetrisko