black-parrot/black-parrot #406: Examine synthesis output of BHT

taylor-bsg, 2020-03-24:
https://github.com/black-parrot/black-parrot/blob/9b3cea5c453d307f893044be6cd5c09f907cab9e/bp_fe/src/v/bp_fe_bht.v#L40

Really curious what this mixed idiom code turns into, and the comparison in area/performance versus a more standard 1R1W version that carries the old value down the pipeline and does not re-read before writing.

taylor-bsg, 2020-03-24:
And also vs. using an RF, and also clock gating efficacy.

dpetrisko, 2020-03-24:
To clarify, the experiment is to pass the previous state in the branch metadata, in addition to the index?  And this would allow us to use a sync mem, because we’re not reading and adjusting in the same cycle. 

That is interesting, and there’s a very direct trade-off. Potentially smaller area BHT for larger area fe_queue fifo and dispatch registers (for additional metadata). No performance impact

taylor-bsg, 2020-04-10:
Yeah, and we could potentially have a much bigger BHT, but at the cost of
having less nimble feedback.

M

dpetrisko, 2020-11-17:
Changed to 1R1W memory as part of #662

closed 2020-11-17 by dpetrisko