black-parrot/black-parrot #408: COSIM terminates too early

dpetrisko, 2020-03-24:
When running with COSIM_P=1 DUMP=1, cosim terminates the simulation from CPP. However, because $finish has not been called, the vcd has not necessarily been dumped. This ends up with the unfortunate side effect of losing the few cycles before the cosim mismatch. 

What we really want is the cosim mismatch + 1/2 cycles, so that the user can look at what's going on during the failure.

taylor-bsg, 2020-03-24:
Maybe there is a way for cosim to call through to the finish function first.

dpetrisko, 2020-03-24:
Yep, that's what Farzam and I discussed.  Adding another DPI call from dromajo->verilog to call finish

farzamgl, 2020-03-24:
Yeah. Actually doesn't even need a new DPI call. We can just add a return code to the existing `step` call and call $finish based on that.

kabylkas, 2020-03-25:
I just thought this might be useful. To see the activity post-mismatch, I did this dirty work around in the `dromajo_step()` DPI. This basically commits a couple of more instructions after the mismatch (depending on the value of the `counter`).
```
 54   exit_code = dromajo_cosim_step(dromajo_pointer, hart_id, pc, insn, wdata, 0, check);
...
 60   if (exit_code != 0) {
 61     kill_soon = true;
 62   }
 63
 64   if (kill_soon) {
 65     if (counter == 0) {
 66       std::cout << "oops!" << std::endl;
 67       abort();
 68     }
 69     counter--;
 70   }
```

dpetrisko, 2020-03-25:
Thanks @kabylkas ! Definitely a reasonable workaround.  It's better for us long term to call $finish because simulators e.g. VCS do some bookkeeping after finish such as flushing buffers, closing files, etc. It's possible that the counter value needs to be tool, system or even system load dependent

closed 2020-03-29 by farzamgl