black-parrot/black-parrot #410: Instruction trace mismatches

dpetrisko, 2020-03-25:
Some benchmarks are showing different instruction traces when changing microarchitectural parameters.  For example, executing rsort on bp_softcore with fe_queue_els_p 8->16 results in the following difference:

```
[CORE0 FSH] PASS
[CORE0 STATS]
	clk   :               386819
	instr :               370871
	fe_nop:                    0
	be_nop:                    0
	me_nop:                    0
	poison:                13829
	roll  :                 1272
	mIPC  :                  958

[CORE0 FSH] PASS
[CORE0 STATS]
	clk   :               386762
	instr :               369871
	fe_nop:                    0
	be_nop:                    0
	me_nop:                    0
	poison:                13772
	roll  :                 1272
	mIPC  :                  956
```

Looking at the instruction traces, there are actually differences in the number of instructions.  Both variants pass cosim, so whatever bad instret logic exists is also affecting that

dpetrisko, 2020-04-05:
Fixed in dev.  Problem was poisoned non-instructions were being spuriously counted

closed 2020-04-05 by dpetrisko

taylor-bsg, 2020-04-05:
Yay! Score one for 477 + professor analyzing student reports for oddness.

M