black-parrot/black-parrot #415: Enable Verilator tests in CI

dpetrisko, 2020-04-01:
Verilator tests pass locally, but fail in CI. Would like to track this down someday, but the logs/reporting of our CI jobs make it very difficult.

- Enable better logging / reporting of runs
- Figure out why tests are failing
- Re-enable tests

muwyse, 2020-04-30:
I believe this should be resolved now from: https://github.com/black-parrot/black-parrot/pull/457 and https://github.com/black-parrot/black-parrot/pull/461.

As of https://github.com/black-parrot/black-parrot/commit/cd949112f849edc1a102fdde71375d156fa55798, CI is running with top-riscv-verilator job enabled, and passing: https://gitlab.com/black-parrot/black-parrot/pipelines/141232847

Next step is to try re-enabling verilator jobs for multicore, beebs, and coremark.

dpetrisko, 2020-04-30:
Oh yeah, those all pass.  I ran them manually on the last job, so we can reenable whenever

muwyse, 2020-04-30:
See: https://github.com/black-parrot/black-parrot/pull/465

dpetrisko, 2020-05-03:
fixed by #465

closed 2020-05-03 by dpetrisko