black-parrot/black-parrot #417: Enable easy selection of FSM or Microcode CCE in design

muwyse, 2020-04-02:
Currently, we must apply a change/patch to substitute in the FSM-based CCE for the microcode-based CCE. It would be ideal to enable this substitution without having to apply a code change. A couple possibilities (please suggest others!):

- Add a use_fsm_cce_p param to bp_params_p, and do conditional instantiation in the modules that instantiate the CCE or its nonsynth tracer (testbenches). The pro here is the code compiles as is using whatever the default is. A downside is it potentially doubles the number of configs we have (e.g., e_bp_single_core_cfg and e_bp_single_core_fsm_cce_cfg)

- Use a tick define or some other keyword and do a text substitution via the Makefile, similar to BP_CFG_FLOWVAR. Pro is we don't need the double configs, con is the design would require this preprocessing step before compilation. Something like this works fine for testbenches, but is probably not ideal for portability/packaging of the design source.

- others?

dpetrisko, 2020-04-02:
We could do 2 without the sed step. In aviary defines, we could do:

‘ifndef BP_CCE
‘define BP_CCE bp_cce
‘endif

And then compile with +define+BP_CCE=bp_cce_fsm to switch it out. 

The downside to this is that it globally defines the CCE as fsm or ucode, so it’s impossible to mix and match. Is that important? Who’s to say

muwyse, 2020-04-02:
ah, that could work well.

I don't think there is a need for mix and match at the moment. Since memory is striped across the CCE's there probably isn't benefit to having only some of them be programmable. The ucode CCE wouldn't be able to do a different coherence protocol on the fly unless the FSM CCE also supported runtime switching of the protocol, which is not planned.

taylor-bsg, 2020-04-02:
I think it depends on whether we expect others to use the flag. If it is a
supported feature of BP, it should be a config param. Otherwise it could be
preprocessor hackery, but usually we reserve that for when we are trying to
factor something in a way not supported by SV built in meta programming
support, but it seems like SV’s meta programming support is sufficient here?

dpetrisko, 2020-04-02:
Yeah that’s reasonable. I think there’s a use case for the FSM CCE as an FPGA build (since there is less motivation for post-synth configurability).  I’m sold on it being a parameter. 

From the perspective of code reduction, we could have a CCE wrapper which instantiates whichever module we want? It’s not a super big savings though, so the if statement could be fine

muwyse, 2020-04-03:
I think there is also a strong case for having the FSM CCE for adoption. I suspect many users would rather just have an FSM CCE than deal with the microcode CCE, for now.

I agree a parameter is the better choice. Using a wrapper may be beneficial since it would unify the modules to a common name, allowing the CCE tracer to bind to the wrapper, for example (makes testbenches less complex). The wrapper would be strictly passthrough and only do the conditional instantiation.

closed 2020-04-07 by muwyse