black-parrot/black-parrot #425: ucode CCE should be able to start in cached mode

dpetrisko, 2020-04-05:
The ucode CCE currently needs to start in uncached mode in order to cache the transition to cached mode when the config bus changes.  In order to support static config bus configurations (minimal hardware), it should be able to handle a static icache_mode = cached, dcache_mode = cached, cce_mode = cached configuration.

muwyse, 2020-04-06:
My understanding was this is an issue with the FSM CCE, not the ucode CCE, per this discussion: https://github.com/black-parrot/black-parrot/pull/416#discussion_r403595999. @sripathi-muralitharan or @Shashank-Vijay, can you confirm this is an FSM CCE issue?

The ucode CCE waits for the mode to transition to normal before fetching instructions. Until that happens, there is a state machine in the message unit that can send and receive uncached requests.

There appears to be a bug in the FSM CCE where it skips the Sync/Ack sequence if the mode is normal at startup. It can be fixed by changing the word "READY" to "SEND_SYNC" on this line:
https://github.com/black-parrot/black-parrot/blob/dev/bp_me/src/v/cce/bp_cce_fsm.v#L748

If PR #416 will be merged soon, we can roll the fix into that. If not I can push a fix to dev.

dpetrisko, 2020-04-06:
My bad. I think rolling this fix into the TB branch is fine.  

BTW, what are the restrictions on mode switches that we have? Is it allowed to switch CCEs before LCEs, can we switch from cached to uncached mode, etc? We should document these behaviors

muwyse, 2020-04-06:
Both CCE's begin operating as soon as reset goes low.

ucode (freeze is never considered by the ucode CCE):
- CCE clears its directory state
- until mode==normal gets set, a state machine runs that processes uncached load/store requests from the LCE Request port, issues uncached load/store to memory, and consumes the uncached read/write response from memory, forwarding as uncached load data or uncached store done to requesting LCE.
- at transition time from uncached to cached, the uncached state machine processes any remaining outstanding uncached responses from memory before allowing the microcode to process messages
- the ucode CCE begins fetching instructions when mode changes to normal. The message unit stalls instruction execution until outstanding uncached accesses are resolved.
- ucode CCE does what the ucode says, currently the first routine is the LCE Sync/Ack sequence

FSM (freeze must be low to enter normal mode):
- the FSM CCE starts up by clearing the directory state
- CCE enters either an uncached mode or normal mode
- the transition to normal mode occurs only if freeze is low AND mode is normal
- if normal mode is not entered directly after clearing the directory, the FSM CCE enters an uncached only mode
- when normal mode is entered (post bug fix), it goes to the LCE Sync/Ack sequence.
- NOTE: gating on freeze should probably be removed from the FSM CCE, so it transitions to normal mode as soon as the mode bit on the CFG bus is set, to be consistent with the ucode CCE behavior.

I think the CCE needs to switch to normal mode prior to the LCEs switching modes, otherwise the CCE probably won't handle the request properly.

Changing modes from normal to uncached only is not supported.

closed 2020-04-07 by muwyse

dpetrisko, 2020-04-07:
Thanks! I made a note for myself to write this up in the docs