black-parrot/black-parrot #426: Support baremetal command line arguments for tests

dpetrisko, 2020-04-05:
Lots of ways to do this.  Ideally, we would be able to pass in arguments from the verilog testbench, so that we don't have to recompile code for each system configuration.

Options off the top of my head:
- add plusargs to nonsynth_host, store in registers, request these MM-registers from start.S
- elf hackery during baremetal elf load
- Another round of linking right before simulation
- could just compile -D if the compilation is not a big deal

muwyse, 2021-04-29:
What do you mean by "baremetal command line arguments"?

dpetrisko, 2021-04-29:
@sripathi-muralitharan is working on this, but basically letting us doing something like:

make sim.v SUITE=bp-tests PROG=mc_sanity CARGS=8

sripathi-muralitharan, 2021-04-29:
I am working on something that is quite manycore specific but we can just modify that to use it with our system as well. Basically, VCS allows you to push in arguments via the +c_args option. We can capture that in the host (maybe with some C/C++ code in the host - that's how the manycore does it) and then have our start code query that from the host through uncached accesses and populate a array which the program running on BP could use.

dpetrisko, 2021-04-29:
I don’t think it’s manycore specific...