black-parrot/black-parrot #432: What is a DMS file

CalebKierum, 2020-04-08:
The getting started slides (on the README.md) link to [this file currently which is a DMS file ](https://fosdem.org/2020/schedule/event/riscv_blackparrot/attachments/slides/3718/export/events/attachments/riscv_blackparrot/slides/3718/Talk_Slides). I am not entirely sure what this is or how to open it. I have managed to unarchive it to a folder of tiffs but am guessing it is some sort of powerpoint equivalent software if these are indeed the slides. Would it be possible to link the slides as a PDF or perhaps give instructions on how to open linked file?

dpetrisko, 2020-04-08:
Hi Caleb, thanks for the issue!

Interestingly, I never noticed this because Preview on macOS Catalina just opens this file natively (but could not figure out how to open it on my phone).  I'll make sure to add a better link. Here's a link to the pdf in the meantime:  [pdf link](https://drive.google.com/file/d/16BXCT1kK3gQ0XKfZPR-K8Zs2E648qFp9/view?usp=sharing )

dpetrisko, 2020-11-21:
Closing with #685

closed 2020-11-21 by dpetrisko