black-parrot/black-parrot #439: CI not reporting fail for regress_riscv.v properly

muwyse, 2020-04-12:
Without fully specifying both TB and CFG parameters in the gitlab CI file, some tests were running with TB=bp_processor and CFG=e_bp_softcore_cfg. I don't believe this is intended to be a supported TB/CFG pair.

Notably, the TB/CFG pair of bp_processor & e_bp_softcore_cfg results in the regress_riscv.v target failing some tests, but this does not appear to be detected by CI. Specifically, the rv64ui-v-* tests and rv64ui-p-fence_i tests report failures when I run them locally. Changing CFG to e_bp_single_core[_ucode_cce]_cfg results in all tests passing.

This pipeline: https://gitlab.com/black-parrot/black-parrot/pipelines/135024242 shows the CI tests reporting pass at a commit where there are failing tests due to the TB/CFG mismatch. Commit is 22579ef (current head of dev as of 4/12/20).

dpetrisko, 2020-04-12:
Interesting. Do you have an idea why CI isn’t picking this up?

muwyse, 2020-04-12:
I think it is the bash script that runs the job and the "grep -cr" command. I think we might need to do "grep FAIL | wc -l" and then check if that returns > 0 (fail) or == 0 (pass).

dpetrisko, 2020-04-12:
Yeah, that’s what it’s doing. I think it might be if the job dies instead of failing, it may not even make a report file to grep

muwyse, 2020-04-12:
I don't think the grep -cr command is working as we expect. If you do the steps below, individual tests will fail, but you can see in the CI pipeline linked above that it is reporting the test passes.

To repro:
- check out "dev" at 22579ef
- cd bp_top/syn
- make build.v TB=bp_processor
- make regress_riscv.v TB=bp_processor -j 16

muwyse, 2020-04-12:
> I think it might be if the job dies instead of failing, it may not even make a report file to grep

This is likely true, too. But the case above doesn't have any dying jobs, they all run to completion, but some fail.

muwyse, 2020-04-30:
Fixed with: https://github.com/black-parrot/black-parrot/pull/457 and https://github.com/black-parrot/black-parrot/pull/461
Closing.

closed 2020-04-30 by muwyse