black-parrot/black-parrot #444: Getting Started (full): No Mac instructions

CalebKierum, 2020-04-16:
It would be incredibly useful to know what the Mac equivalents are for these initial commands as it does not have yum or apt-get out of the box. Homebrew (brew) worked for most of the installations except for libmpc-devel mpfr-devel gmp-devel, gcc-c++, zlib-devel , expat-devel , and gtkwave. That being said brew does know of libmpc, mpfr, gmp, zlib, and expat but I do not know if those will work just as well.

dpetrisko, 2020-04-16:
Thanks for the issue! We don't really test on Mac, since CAD tools tend to run on CentOS and less Ubuntu.  That said, it would be great to run on Mac for the open-source toolchain (gcc+verilator). If anyone manages to get it working, we'd love a PR.

dpetrisko, 2022-06-27:
Closing as won't fix. We support CentOS 7 and strive for but do not promise Ubuntu support. Mac users are encouraged to use Docker

closed 2022-06-27 by dpetrisko