black-parrot/black-parrot #445: Make of libdramsim.so fails due to potential linking issue "soname"

CalebKierum, 2020-04-16:
This is most likely my own system's configuration however I am having difficulty getting the build scripts to work. It appears to happen during the make of libram `make -C /Users/ckierum/Coding/ComputerArchitecture/black-parrot/external/DRAMSim2 libdramsim.so` and the specific error is `ld: unknown option: -soname` which is probably triggered by this final command prior to the error `g++ -g -shared -Wl,-soname,libdramsim.so -o libdramsim.so AddressMapping.po Bank.po BankState.po BusPacket.po ClockDomain.po CommandQueue.po IniReader.po MemoryController.po MemorySystem.po MultiChannelMemorySystem.po PrintMacros.po Rank.po SimulatorObject.po Transaction.po`

I looked into this a bit and it could potentially be linked to the question on [this Stack Overflow ](https://stackoverflow.com/questions/4580789/ld-unknown-option-soname-on-os-x) post but I am not sure as they suggest what appears to be changes to build scripts so I wasn't sure if I should follow this. 

I have attached the full build log.
[failedLibrary.log](https://github.com/black-parrot/black-parrot/files/4488577/failedLibrary.log)

dpetrisko, 2020-04-16:
Yes, I remember running into this issue trying to get this to run on my machine.  IIRC, I ended up following that SO post.  We'd probably want to do something like: https://gist.github.com/sighingnow/deee806603ec9274fd47 to support both parameter names

dpetrisko, 2022-06-27:
Feel free to reopen if still relevant for others

closed 2022-06-27 by dpetrisko