black-parrot/black-parrot #459: beebs: levenshtein failing in regressions

muwyse, 2020-04-29:
Levenshtein benchmark in beebs suite is failing. Stall detected for Core 0.

At commit: https://github.com/black-parrot/black-parrot/commit/b777f7d81ddf0266ad3ea6b44ce2d5f2039b8f48

Repro:
cd bp_top/syn
make build.v
make regress_beebs.v -j 20

Note: fails for both bp_softcore (CFG=e_bp_softcore_cfg) and bp_processor (CFG=e_bp_single_core_cfg) testbenches.

dpetrisko, 2020-04-29:
```  
   80001140:   03a605b3            mul a1,a2,s10
    80001144:   f4043603            ld  a2,-192(s0)
    80001148:   03a63633            mulhu   a2,a2,s10
```

Problem is this.  We trap and emulate mulhu, but our trap tables aren't set up here.  We should be linking libperch with beebs, but I guess we're not? @farzamgl ?

farzamgl, 2020-04-29:
No, we're not. I tried to link then with perch but they use a weird config file approach that just didn't link the library no matter what.
The bp files can be found under `beebs/config/riscv64/chips/black-parrot`.

dpetrisko, 2020-04-29:
Oh weird.  What do you think is the easiest way to solve this?

farzamgl, 2020-04-29:
Easiest is to add the files in that repo and include them in the chip.cfg file. I have no idea how to link the perch though.

dpetrisko, 2020-04-29:
Fixed in #461

PR #461 (2020-04-29): Bumping beebs
fixes #459
closed 2020-04-29 by dpetrisko