black-parrot/black-parrot #460: Vivado synthesis issue with dcache

dpetrisko, 2020-04-29:
ERROR: [Synth 8-1587] an enum variable may only be assigned to same enum typed variable or one of its values [/home/petrisko/scratch/bsg_fpga/vu37p/black-parrot/bp_be/src/v/bp_be_mem/bp_be_dcache/bp_be_dcache.v:989

@muwyse, I think you have the last changes here

muwyse, 2020-04-29:
Looks like it just needs a cast to bp_coh_states_e

dpetrisko, 2020-04-29:
Oh, I think you’re right.  I thought it was the replication operator. Usually VCS flags assigning bits to an enum, not sure why it made it to vivado

muwyse, 2020-04-29:
I'm not sure either. Neither the VCS or Verilator linter seems to catch it either

muwyse, 2020-04-30:
Fixed with: https://github.com/black-parrot/black-parrot/pull/462
Closing.

closed 2020-04-30 by muwyse