black-parrot/black-parrot #474: Cosim mismatch on integer divide instructions

kabylkas, 2020-05-05:
I have several random tests failing cosim on `div` instruction:

### Test 1
```
0 0x000000000000030c (0x0308fcbb) x25 0x0000000000000000 remuw   s9, a7, a6
0 0x0000000000000310 (0x000b08bb) x17 0xffffffff8003dcf8 addw    a7, s6, zero
0 0x0000000000000314 (0x032d8f3b) x30 0x0000000000000000 mulw    t5, s11, s2
0 0x0000000000000318 (0x021fca3b) x20 0x0000000000000008 divw    s4, t6, ra
[error] EMU PC 0000000000000318, DUT PC 0000000000000318
[error] EMU INSN 021fca3b, DUT INSN 021fca3b
[error] EMU WDATA 0000000000000008, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00140080, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
COSIM_FAIL
```

if you trace back the registers, values are: `t6 = 0xdaa11633` and `ra = 0xfb55caa5`. `divw` is a 32 bit signed division so the result should indeed be 8. The result is zero if you do unsigned div.

Binary: [riscv_illegal_instr_test_1.S.riscv.zip](https://github.com/black-parrot/black-parrot/files/4578390/riscv_illegal_instr_test_1.S.riscv.zip)

### Test 2
```
3 0x00000000800001dc (0xff3d8e9b) x29 0xfffffffffffffff3 addiw   t4, s11, -13 
3 0x00000000800001e0 (0x033bff3b) x30 0x0000000000000000 remuw   t5, s7, s3
3 0x00000000800001e4 (0xd64b8c1b) x24 0xfffffffffffff9d0 addiw   s8, s7, -668
3 0x00000000800001e8 (0x0289c9bb) x19 0xffffffffffffffff divw    s3, s3, s0
[error] EMU PC 00000000800001e8, DUT PC 00000000800001e8
[error] EMU INSN 0289c9bb, DUT INSN 0289c9bb
[error] EMU WDATA ffffffffffffffff, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00100080, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
COSIM_FAIL
```
* `s3 = 0x0000000000000001`
* `s0 = 0x8000000000000000`

This wasn't clear to me :smiley: `s0` is a huge negative number, does the result should be negative?

Binary: [riscv_arithmetic_basic_test_0.S.riscv.zip](https://github.com/black-parrot/black-parrot/files/4578424/riscv_arithmetic_basic_test_0.S.riscv.zip)

I have several other but it looks like the same issue.

dpetrisko, 2020-05-05:
Thanks for the report!

Great catch, I believe this is the bug(/fix):
https://github.com/black-parrot/black-parrot/commit/d6d777f532e850f4cb84c7aa5aa9552c35b8d5cb

taylor-bsg, 2020-05-05:
> d6d777f

Dan can we add an assembly test to the regression for this bug? (NB I never
use the signed keyword in SV because of how many bugs result from its use.)

M

kabylkas, 2020-05-08:
Thank you, Dan. Your patch fixed the issue.

closed 2020-05-08 by kabylkas