black-parrot/black-parrot #475: UCE writeback sequence should wait for the write response

farzamgl, 2020-05-05:
Currently, during a writeback, UCE sends a read commands followed by a write command to writeback the dirty data. But after sending these commands it only waits for the read response before jumping back to the ready state. This could cause the late write response from the cache to happen at the same time as the response for the next instruction and break the top-level arbiter between cache and other devices.

dpetrisko, 2020-05-05:
We shouldn't wait on writebacks (performance hit) but we should fix the response arbitration.

The sequence here is:
- WB read
- WB write
- IO read

- WB read response
- WB write response / IO read response

We should write a basic test for this, also

farzamgl, 2020-05-05:
What's the point of the wb write response? Where is it used?

dpetrisko, 2020-05-05:
credit counting for fences

taylor-bsg, 2020-05-05:
Seems like a warning sign about this module that there is some concurrency
that has not been thought through?

dpetrisko, 2020-05-05:
Yeah, the problem is that responses from different destinations can return unexpectedly. The arbitration was written as one-hot because it was assumed there was only 1 outstanding request from each cache, but this does not hold when CLINT or I/O is involved

taylor-bsg, 2020-05-05:
Sounds like in the comments there should be some explicit reasoning about
what concurrent events can happen. Arbiters dropping things under rare
concurrent events is common bug pattern that takes a long time to track
down due to the infrequency of such events.  Generally the assumption for
any module should be that if it has multiple interfaces, they are
concurrent; and when possible non-concurrent interfaces should be collapsed
into a single interface (particularly if the non-concurrent interface comes
from the same place).

Will update the style guide with this...

dpetrisko, 2020-05-06:
The bug isn't in the UCE.  The bug is in the arbitration here:
https://github.com/black-parrot/black-parrot/blob/master/bp_top/src/v/bp_softcore.v#L428

Note the (incorrect) comment that says the responses are guaranteed to be one-hot

taylor-bsg, 2020-05-07:
Oh I see. Perhaps we should have a BSG_ASSERT_ONE_HOT(clk,signal) and a
BSG_ASSERT_ONE_HOT_OR_ZERO(clk,signal) in BaseJump STL to check one
hotness? There are some SVA primitives, but presumably we want to have
module-wide control over them for tool independence.

M

PR #488 (2020-05-11): Fixing softcore resp arbitration
There may be multiple simultaneous returning requests to bp softcore from clint, cfg, cache or I/O.  To handle this conservatively, this PR inserts fifos on the response path as well.  It's also possible to fix this by masking the yumis to these devices, but this is a more timing path friendly approach.  If there ends up being a performance issue, we can bypass the fifos.  If there ends up being an area issue, we can reevaluate.

Fixes #475
closed 2020-05-12 by dpetrisko