black-parrot/black-parrot #476: Libperch should contain a way to query platform data

dpetrisko, 2020-05-05:
Such as number of cores, etc.  The values could be set by a linker script or somehow encoded in the hardware.

This could avoid needing to recompile baremetal programs for different configurations like this:

https://github.com/black-parrot/black-parrot/blob/master/bp_common/test/src/demos/src/mc_sanity.c#L17

dpetrisko, 2020-05-05:
Also related to #426

muwyse, 2021-04-29:
Having number of cores and related be exposed as some type of read-only memory-mapped registers would be useful and I think quickly solve the issue of multiple instances of those baremetal mc_* programs.

dpetrisko, 2021-04-29:
Yeah I was thinking of doing something similar to manycore (some memory-mapped configuration ROM that lives in host): https://github.com/bespoke-silicon-group/bsg_replicant/blob/main/hardware/hardware.mk#L151

muwyse, 2021-08-17:
I believe this has been added in perch as of: https://github.com/black-parrot-sdk/perch/pull/5 and https://github.com/black-parrot-sdk/bp-tests/commit/40b26a06d10a217a7d1e398386dbe9f6b4844a40

@dpetrisko ready to close?

closed 2021-08-20 by black-parrot