black-parrot/black-parrot #478: Restore AXE consistency checker targets

dpetrisko, 2020-05-05:
This can help us verify coherence in the multicore cache testbenches and CCE testbenches

muwyse, 2020-05-11:
A couple options:

1. A new testbench in the ME for this. AXE requires all accesses to be the same size (so it can properly track them), and since we are checking consistency, we should really have multiple LCEs. To avoid unnecessary complexity in the existing ME testbench, I think we want the multi-LCE setup to be its own testbench. We also need new configurations, or the ability to override the number of CCE's, since we likley want to test multi-LCE but single CCE.

2. figure out how to strap this on to our existing multi-core configs with the full bp_processor. We would need a test program that only does load and store for single size (i.e., 64-bits), and we would need to force the simulated memory to return 0 in the case of load before store to location A. We could attach an AXE tracer to the D$ for creating axe logs, then some scripts/post-processing to do the consistency check.

muwyse, 2021-08-17:
Update: ME testbench has skeleton code in place to instantiate a multi-LCE, single CCE system. The remaining issue is generating the test ROMs for each LCE and modifying the testbench to ignore the read values returned by the LCE, since we can't statically determine the read value of operations when multiple LCEs are racing to read/write words in memory.

Generating test ROMs should be relatively easy as the infrastructure already supports generating random load/store sequences. It is easily modifiable to constrain the load/store sizes that can be issued by the LCEs.

And AXE needs to be added either to tools/ or the SDK (also easy)

muwyse, 2021-10-08:
Adding two tasks:
- [ ] update ME testbench for multi-LCE with an AXE test (64-bit only load and store)
- [ ] CI script and Makefile updates to automatically run AXE check from existing log

muwyse, 2021-10-08:
Task 1 is being addressed with #944

muwyse, 2021-10-08:
A couple notes:
1. AXE does support AMOs, the syntax for the trace file is <M[0] == 0; M[0] := 1>, which just shows the read followed by the write with the angle brackets grouping them together. Basically, it is a matter of detecting the AMO and capturing the load and store values.
2. LR/SC is handled similarly, where a passing LR/SC sequence should be shown in the trace like an AMO above, and a failing one is shown by converting the LR to a standard load in the trace. This probably takes a little bit of clever tracing, but shouldn't be too bad.
3. Currently, the ME testbench uses the mock LCE, and I don't think I added AMO support to it. Task 3 above will address use of AXE with our top-level setup.