black-parrot/black-parrot #483: ME regression passes in VCS, but not Verilator

dpetrisko, 2020-05-07:
2eaabec0 in sw_dev.

```make -C bp_me build.v sim.v``` passes

```make -C bp_me build.sc sim.sc``` fails with trace mismatch

I think this may have snuck in with the new Verilator version.  If that's the case, we can talk about reverting it, but it would be nice to include the latest build system changes they've committed.  I've anecdotally found that when VCS works, but Verilator doesn't, it tends to be non-compliant SV code, which would be nice to root out

muwyse, 2020-05-12:
Very puzzling. I ran both and diffed the CCE trace's. The Verilator one simply has fewer lines, but all the lines it contains match exactly with the VCS trace.

The LCE traces show that in the Verilator simulation a store appears to be dropped by the LCE. Still investigating, but it is almost certainly a bug in the mock LCE. My guess is there is a logic bug with committing the store after getting coherence permissions for the block, or the nonsynth nature of the mock LCE is causing problems.

The upside here is it appears to be a bug with the nonsynth mock LCE, not the actual processor or CCE.

muwyse, 2020-05-30:
Root cause appears to be a regression in Verilator tool itself related to bit selection from a 2D array/vector. Mock LCE has been updated to use different syntax to avoid, with comment about problem.

Fixed by #523

closed 2020-05-30 by muwyse