black-parrot/black-parrot #485: Building BP with Docker Ubuntu

chaserwhytey, 2020-05-07:
I followed the instructions to build BP with Docker, and once I got to the step where you're supposed to run make prep_lite, in the comments it says you can run **make prep_lite -j** for faster builds. However, I received the error below when I tried this. When I ran make prep_lite without the -j, it worked just fine. Not sure if it is a problem on my end or on BP end.

 CXX      utils/sc_mempool.lo
  CXX      utils/sc_pq.lo
  CXX      utils/sc_report.lo
  CXX      utils/sc_report_handler.lo
  CXX      utils/sc_stop_here.lo
  CXX      utils/sc_string.lo
  CXX      utils/sc_utils_ids.lo
  CXX      utils/sc_vector.lo
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/int/sc_signed.lo' failed
make[7]: *** [datatypes/int/sc_signed.lo] Error 1
make[7]: *** Waiting for unfinished jobs....
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/bit/sc_bv_base.lo' failed
make[7]: *** [datatypes/bit/sc_bv_base.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'tracing/sc_wif_trace.lo' failed
make[7]: *** [tracing/sc_wif_trace.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/int/sc_unsigned.lo' failed
make[7]: *** [datatypes/int/sc_unsigned.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/int/sc_int_base.lo' failed
make[7]: *** [datatypes/int/sc_int_base.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/int/sc_uint_base.lo' failed
make[7]: *** [datatypes/int/sc_uint_base.lo] Error 1
Makefile:1072: recipe for target 'communication/sc_signal_ports.lo' failed
make[7]: *** [communication/sc_signal_ports.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'kernel/sc_phase_callback_registry.lo' failed
make[7]: *** [kernel/sc_phase_callback_registry.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/bit/sc_lv_base.lo' failed
make[7]: *** [datatypes/bit/sc_lv_base.lo] Error 1
Makefile:1072: recipe for target 'datatypes/fx/sc_fxnum.lo' failed
make[7]: *** [datatypes/fx/sc_fxnum.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'kernel/sc_module.lo' failed
make[7]: *** [kernel/sc_module.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/misc/sc_concatref.lo' failed
make[7]: *** [datatypes/misc/sc_concatref.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'datatypes/fx/sc_fxval.lo' failed
make[7]: *** [datatypes/fx/sc_fxval.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'tracing/sc_vcd_trace.lo' failed
make[7]: *** [tracing/sc_vcd_trace.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'communication/sc_prim_channel.lo' failed
make[7]: *** [communication/sc_prim_channel.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'utils/sc_vector.lo' failed
make[7]: *** [utils/sc_vector.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'kernel/sc_process.lo' failed
make[7]: *** [kernel/sc_process.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'communication/sc_event_queue.lo' failed
make[7]: *** [communication/sc_event_queue.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'communication/sc_clock.lo' failed
make[7]: *** [communication/sc_clock.lo] Error 1
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1072: recipe for target 'communication/sc_signal_resolved_ports.lo' failed
make[7]: *** [communication/sc_signal_resolved_ports.lo] Error 1
make[7]: Leaving directory '/home/build/black-parrot/external/systemc/src/sysc'
Makefile:1124: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/build/black-parrot/external/systemc/src/sysc'
Makefile:654: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/build/black-parrot/external/systemc/src'
Makefile:516: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/build/black-parrot/external/systemc'
/home/build/black-parrot/external/Makefile.tools:32: recipe for target 'systemc_build' failed
make[3]: *** [systemc_build] Error 2
make[3]: Leaving directory '/home/build/black-parrot'
/home/build/black-parrot/external/Makefile.tools:104: recipe for target '/home/build/black-parrot/external/touchfiles/systemc.86e22d96acf8c22e727cddfd33adecf5258e0a5d' failed
make[2]: *** [/home/build/black-parrot/external/touchfiles/systemc.86e22d96acf8c22e727cddfd33adecf5258e0a5d] Error 2
make[2]: Leaving directory '/home/build/black-parrot'
/home/build/black-parrot/external/Makefile.tools:105: recipe for target '/home/build/black-parrot/external/touchfiles/verilator.c2b49f01745ec22ddd96f1fd72faa4ba3ca3d056' failed
make[1]: *** [/home/build/black-parrot/external/touchfiles/verilator.c2b49f01745ec22ddd96f1fd72faa4ba3ca3d056] Error 2
make[1]: Leaving directory '/home/build/black-parrot'
Makefile:17: recipe for target 'prep_lite' failed
make: *** [prep_lite] Error 2

dpetrisko, 2020-05-08:
Thanks for the report! Most likely a Docker / environment issue. If you’re on Ubuntu, I would recommend just building natively unless there’s some strong reason to use Docker.

chaserwhytey, 2020-05-08:
No problem. There is actually a strong reason to use Docker. I'm running on a Mac, and I've heard that there are problems with setting it up on a Mac. Anyways got it up and running, thanks!

dpetrisko, 2020-05-08:
Oh, I misunderstood you! I thought you were running Docker on Ubuntu, not an Ubuntu Docker image.  Which version of MacOS/Docker are you running?

chaserwhytey, 2020-05-08:
MacOS Catalina version 10.15.4
Docker version 19.03.8

developandplay, 2021-03-24:
@dpetrisko 
This error also occasionally occurs for me on Ubuntu without docker for any of the make commands.
I think it's probably due to high resource usage conflicting with the operating system.
Initially I thought I could specify the number of cores like for verilator. However it seems like it's not possible or I'm missing something.
Currently I either have 10 minutes of build time or 1 minute with the risk crashing.

dpetrisko, 2021-03-24:
There's nothing too fancy going on here, it's just multithreaded GCC commands for the most part. Are you on a VM or native install?

developandplay, 2021-03-24:
Just a regular native install.
I mean I don't think there is anything wrong with using all threads if you have a vm or dedicated server.
It would be nice to set it if you run it on a native install.

dpetrisko, 2021-03-24:
>all threads if you have a vm 

In an ideal world, but I've seen buggy VMs.

>It would be nice to set it if you run it on a native install.

I'm a bit confused. Is the issue only coming up when you run with all threads? Of course you can use make -j <number of threads> to taste.

developandplay, 2021-03-24:
Ok that's what I thought initially but looking at my resource monitor if I just run `-j` without specifying the number of threads my 16 threads will all be utilized at 100%. If I specify `-j 16` only a few threads will be pushed to 100% at a time.
I'll read up some more how `-j` works as this is really confusing me.

dpetrisko, 2021-03-24:
Optimal makefile threads depend on a lot of factors: core count, SMT, available memory... Generally accepted wisdom is to use [number of logical cores]+1 as a default (to prevent blocking on I/O threads).

Fun exercise, but we should be running prep pretty rarely, so I wouldn't worry about it too much.

developandplay, 2021-03-24:
Actually this occured for me both on `prep` and just regular `build.sc`.
Anyways apparently j is not the number of threads/cores but actually just "jobs". So if I don't specify the number `make` will just spam the CPU with jobs (risking a crash) and it's just how make is designed. So I'll just go with your suggestion to get a modest speedup. Thanks!

dpetrisko, 2022-06-27:
Closing as I think the root cause here was the usage of unrestricted -j

closed 2022-06-27 by dpetrisko