black-parrot/black-parrot #487: Cosim mismatch on load instruction

kabylkas, 2020-05-09:
### Summary
For one of the random tests there is a mismatch when dromajo and black parrot are executing load instruction:
```c
3 0x0000000080012444 (0x0e04be03) x28 0x0000000080000000 ld      t3, 224(s1)
3 0x0000000080012448 (0x0e84be83) x29 0x000000000000000e ld      t4, 232(s1)
3 0x000000008001244c (0x0f04bf03) x30 0x0000000000000000 ld      t5, 240(s1)
3 0x0000000080012450 (0x0f84bf83) x31 0x000000007704f42b ld      t6, 248(s1)
[error] EMU PC 0000000080012450, DUT PC 0000000080012450
[error] EMU INSN 0f84bf83, DUT INSN 0f84bf83
[error] EMU WDATA 000000007704f42b, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00101800, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
COSIM_FAIL
```

Binary: [riscv_ebreak_debug_mode_test_0.S.zip](https://github.com/black-parrot/black-parrot/files/4602342/riscv_ebreak_debug_mode_test_0.S.zip)
Full dromajo trace: [trace.txt](https://github.com/black-parrot/black-parrot/files/4602366/trace.txt)

⚠️ **Note** ⚠️ : for this test to work(proceed from last cosim fail) I had to fix #400, based on your comment in #402. All I did was masking out LSB of `br_tgt_o`.

### Some more details
Dromajo is loading `0x000000007704f42b`, whereas black parrot is loading zero.

What is happening, I believe, is a classical saving and restoration of registers through stores and loads. If you trace back the requests to the memory, there was a store with that particular value to that address:
1) Previous ld loaded 0x7704f42b to t6 register
```c
3 0x00000000800123b8 (0x0f84bf83) x31 0x000000007704f42b ld      t6, 248(s1)
```

2) Some address manipulation
```c
3 0x0000000080012004 (0x000704b3) x9  0x000000008002ecf8 add     s1, a4, zero
3 0x0000000080012008 (0xf0848493) x9  0x000000008002ec00 addi    s1, s1, -248 
```

3) Series of stores
```c
3 0x0000000080012064 (0x0b74bc23)                        sd      s7, 184(s1)
3 0x0000000080012068 (0x0d84b023)                        sd      s8, 192(s1)
3 0x000000008001206c (0x0d94b423)                        sd      s9, 200(s1)
3 0x0000000080012070 (0x0da4b823)                        sd      s10, 208(s1)
3 0x0000000080012074 (0x0db4bc23)                        sd      s11, 216(s1)
3 0x0000000080012078 (0x0fc4b023)                        sd      t3, 224(s1)
3 0x000000008001207c (0x0fd4b423)                        sd      t4, 232(s1)
3 0x0000000080012080 (0x0fe4b823)                        sd      t5, 240(s1)
// storing 0x000000007704f42b to the address of interest
3 0x0000000080012084 (0x0ff4bc23)                        sd      t6, 248(s1) // <<<
```

4) Later cosim fails when trying to load from that memory address
```c
3 0x0000000080012450 (0x0f84bf83) x31 0x000000007704f42b ld      t6, 248(s1)
[error] EMU PC 0000000080012450, DUT PC 0000000080012450
[error] EMU INSN 0f84bf83, DUT INSN 0f84bf83
[error] EMU WDATA 000000007704f42b, DUT WDATA 0000000000000000
[error] EMU MSTATUS a00101800, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
COSIM_FAIL
```

dpetrisko, 2020-05-09:
Thanks for another excellent report!

Which configuration of BlackParrot are you using? Softcore? With/without L2?

kabylkas, 2020-05-09:
I didn't have chance to play with configurations. Whatever is the default configuration, that's what I used. I believe it is softcore with l2 enabled (e_bp_softcore_cfg).

dpetrisko, 2020-05-10:
Hi Nursultan!

I believe I have found the bug. Try applying #490

kabylkas, 2020-05-11:
Thank you! :pray:

closed 2020-05-11 by kabylkas