black-parrot/black-parrot #497: Cosim mismatch: black-parrot is not trapping on an illegal instruction

kabylkas, 2020-05-13:
### Summary 
For some reason Black-Parrot is trying to commit an illegal instruction instead of trapping, even though there were previous successful traps on illegal instruction before this one.

Instruction at PC `0x800070fc` is an illegal one (`0xfdc9a1e7`). 
```c
...
3 0x00000000800070f4 (0x03cbecbb) x25 0xffffffffffffff2a DASM(0x03cbecbb)
3 0x00000000800070f8 (0x0253c23b) x4  0xffffffffffffffff DASM(0x0253c23b)
3 0x0000000080012000 (0x340494f3) x9  0x000000008002fcf8 DASM(0x340494f3)
[error] EMU PC 0000000080012000, DUT PC 00000000800070fc
[error] EMU INSN 340494f3, DUT INSN fdc9a1e7
[error] EMU WDATA 000000008002fcf8, DUT WDATA 0000000080007100
[error] EMU MSTATUS a00101800, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1
COSIM_FAIL
$finish called from file "/mada/users/nkabylka/black-parrot/bp_top/test/common/bp_nonsynth_cosim.v", line 121.
$finish at simulation time                87680
           V C S   S i m u l a t i o n   R e p o r t 
Time: 87680 ps
``` 
Binary: [riscv_illegal_instr_test_0.S.riscv.zip](https://github.com/black-parrot/black-parrot/files/4624781/riscv_illegal_instr_test_0.S.riscv.zip)
Cosim full trace: [illegal_inst_trace.txt](https://github.com/black-parrot/black-parrot/files/4624767/illegal_inst_trace.txt)
Dromajo stand alone run trace: [trace.txt](https://github.com/black-parrot/black-parrot/files/4624787/trace.txt) (this is to view instructions that through exception)

### Previous successful traps
Before this mismatch, there was a series of illegal instruction occurrences where Black-Parrot did trap: 

1) at PC `0x80006e28` instruction `0x31a4fa5f`
```c
3 0x0000000080006e20 (0x018e2323)                        DASM(0x018e2323)
3 0x0000000080006e24 (0x00be0423)                        DASM(0x00be0423)
3 0x0000000080012000 (0x340494f3) x9  0x000000008002fcf8 DASM(0x340494f3)
3 0x0000000080012004 (0x000704b3) x9  0x000000008002fcf8 DASM(0x000704b3)
```

2) at PC `0x80006dc4` instruction `0x00006681`
```c
3 0x0000000080006dbc (0x004e0023)                        DASM(0x004e0023)
3 0x0000000080006dc0 (0x40025b1b) x22 0x0000000000000078 DASM(0x40025b1b)
3 0x0000000080012000 (0x340494f3) x9  0x000000008002fcf8 DASM(0x340494f3)
3 0x0000000080012004 (0x000704b3) x9  0x000000008002fcf8 DASM(0x000704b3)
```

dpetrisko, 2020-05-13:
Thanks! Confirmed bug.

[image]

RISC-V defines that funct3 for JALR == 3'b000.  In our decoder, we do not make this assertion and just check the opcode.

```
      `RV64_JALR_OP :
        begin
          decode.pipe_ctrl_v = 1'b1;
          decode.irf_w_v    = 1'b1;
          decode.fu_op      = e_ctrl_op_jalr;
          decode.baddr_sel  = e_baddr_is_rs1;
        end
```

dpetrisko, 2020-05-14:
Check out #499 .  I'll leave the issue open because it's possible there are more illegal instruction cases which will be detected by this test.  Those aren't ordinarily executed, so they tend not to be well-tested

dpetrisko, 2020-11-04:
Illegal instruction space is fairly well tested now, passing RISCV-DV tests. Will open new issues if any discovered

PR #499 (2020-05-16): Detecting illegal jalr
[image]

RISC-V defines that funct3 for JALR == 3'b000.  In our decoder, we did not make this assertion and just checked the opcode.

```
      `RV64_JALR_OP :
        begin
          decode.pipe_ctrl_v = 1'b1;
          decode.irf_w_v    = 1'b1;
          decode.fu_op      = e_ctrl_op_jalr;
          decode.baddr_sel  = e_baddr_is_rs1;
        end
```

Fixes #497 , test added
closed 2020-11-04 by dpetrisko