black-parrot/black-parrot #498: Can I randomly toggle ready_o signal? (verification experiment)

kabylkas, 2020-05-13:
Hi guys! I have a quick question. How stable do you think is black-parrot if I start randomly toggling `ready_o` signals across the design? By toggling I mean doing `ready_o & random_signal`. In other words, saying I am not ready even if I am, not vice versa. 

For example, I randomly toggled `ready_o` of FE cmd fifo. My assumption is that this is simply a handshake signal. Doing this will obviously affect performance but shouldn't affect the functionality, am I right? What I am trying to do is to create artificial back pressure to see if this will activate any logic that wasn't active before or bring the processor in a potential buggy state. 

It seems like it does activate something that wasn't active before because all of the tests that I have start failing cosim. I was wondering if what I am doing is legal at all? 

Thank you!
[image]

dpetrisko, 2020-05-13:
Great question and interesting project.  Seems super valuable for stressing latency-insensitive interfaces!  There are a few cases here, and some terminology from the basejump paper (see #327 for a discussion of this) to determine what is legal to toggle.

- ready-valid-and
  - in this case, it is fine to mask ready or valid (1->0).  Raising 1 when not ready or not valid will result in undefined behavior
- ready-then-valid
  - in this case, it is fine to mask ready (1->0).  Rasing 1 on ready or masking 1 on valid will result in undefined behavior
- valid-then-ready (also called valid-then-yumi)
  - in this case, it is fine to mask valid (1->0).  Raising 1 on valid or masking 1 on yumi will result in undefined behavior

Most interfaces in BlackParrot are ready-then-valid or valid-then-yumi, whereas modules in Basejump STL tend to be ready-valid-and.  

That said, it should be legitimate to mask the ready signal of the fifo here.  However, this specific case is actually a known bug (#405 #443).  The BE cannot handle backpressure from the FE in this way, because there is an assumption that the FE will dequeue 1 attaboy a cycle so that the fifo is always ready.

kabylkas, 2020-05-13:
Thank you for an informative response, Dan! Very helpful! I will read through the references that you mentioned

taylor-bsg, 2020-05-14:
I would read the BaseJump STL paper first. Assuming you are injecting the
gates between the modules, you need to mask all helpful signals (which
means for “valid and ready” interfaces you need to mask both signals, a
difference from what Dan said above.)

M

dpetrisko, 2023-07-18:
Closing as stale

closed 2023-07-18 by dpetrisko