black-parrot/black-parrot #503: Certain CSR writes should flush and sync FE

farzamgl, 2020-05-20:
Writing some CSRs like satp and some fields of mstatus that control address translations should flush the FE and sync the shadow translations info. This guarantees that the next instruction after the CSR write can see all the side-effects of that update.

dpetrisko, 2020-05-20:
Which fields of mstatus?

farzamgl, 2020-05-20:
> Which fields of mstatus?

MPRV, for example, can enable translation in M-mode.

dpetrisko, 2020-05-20:
MPRV only affects data, not instructions

farzamgl, 2020-05-20:
You're right. Is the same true for the SUM bit?

dpetrisko, 2020-05-20:
Yep

dpetrisko, 2020-05-20:
[image]

farzamgl, 2020-05-20:
That's all then. I think satp should be enough.

dpetrisko, 2020-05-31:
fixed by #524

closed 2020-05-31 by dpetrisko